Sinapse-Energia/AP-Sinapse #10: [Dummy RF] To develop the necessary SB functions / queues in order to exploit the RF features

ralcaide, 2017-07-31:
As a first step for the RF management development, it should be developed all the wrapper functions of the RF Flow but without sending / receiving really the RF commands. 
See the complete flow diagram:
[RF Flow diagram](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Flow-diagrams#rf-southbound-flow-diagram)

From this flow diagram, it should be developed the following commands:

1. Send_RF_msg(msg, id, args*) . Function to be used by the API engine
3. Set_messages : Save the RF command in a queue

5. Process_RF_SEND_queue : Delete one by one the messages of the queue. Here is where the messages are sent

17. Update Context: Update the context in relation the command received in 1, but when the message is processed in 5. This function is the simulation of the reception of a RF message and its processing

These functions are almost the wrapper needed by the API engine in order to call the RF functions.

francisjjp, 2017-08-07:
For development of functions, the queue procedure is going to be used  in dynamically fashion.

[6 hour]. Development of points 1, 3, 5. Nothing regarding queue reception management and update context.

TODO:  to upload at this point all functions regarding management of TDA5340 transceiver into "RF_Transceiver.h" and "RF_Transceiver.c".

Three functions into "RF_Queue.h" have been tested OK.

francisjjp, 2017-08-07:
[1 hour]. Adding TDA5340 functions to **RF_transceiver.h and RF_transceiver.c**.  Two functions wraps funcionality of one chosen transceiver. In this case, it is used #define TDA5340_transceiver for chosing TDA5340 transceiver.

int Transmit_RF_initialize(void ); /// This function wraps initialize process of chosen transceiver
int Transmit_RF_msg(uint8_t *frame); /// This function wraps transmitting  process of chosen transceiver.

TODO: 
1) Remove spanish words.
2) Wraps de reception part of TDA5340 with one global function (RF_msg_received())
3) Test the working.

ralcaide, 2017-08-14:
@francisjjp , I will integrate this part. 
One question: The RF_RECV_queue is populated in any moment?. When a command is "sent" in the part 5, it should write a command in the RF_RECV_queue

francisjjp, 2017-08-14:
I forgot!. No, it is not populated. OK, for dummy, includes only one called for populating at same time you populates transmit queue.

ralcaide, 2017-08-14:
Please add this process to the dummy:
- When a message is "sent" in 5: To add the "received" command in RF_RECV_queue. 
- 11. Process the RF_RECV_queue: Delete the messages in the queue (Simulate the process: One per iteration).
`while msg in RF_RECV_queue { delete msg, sleep (1) }`

closed 2018-02-08 by ralcaide