Sinapse-Energia/AP-Sinapse #11: [ML1 - Startup] To test / understand the basis version of the AP code

ralcaide, 2017-07-31:
The EasyNet---V0.1 branch contains a first version of the code with a MQTT Client and several commands implemented using a basic context.
TODO

- Test it using the commands 1; and 3;
- Understand how everything is working

fpgamcu, 2017-08-11:
Done testing 2 and 3:
subscribed
`mosquito_sub -h soporte-tecnico.bitnamiapp.com -p 1883 -t LU/LUM/SEN`
`mosquito_sub -h soporte-tecnico.bitnamiapp.com -p 1883 -t LU/LUM/TRACE/999998`
published
`mosquito_pub -h soporte-tecnico.bitnamiapp.com -p 1883 -t LU/LUM/ACT/999998 -m "1;"`
received message in trace and answer in SEN
`999998;36;0;0;300;231;134.0;0.1;148.0;707.0;0.5;690.0;0.0`
that is described in metadata
`"MEASU%$ID;$TEMP;$STAT;$DSTAT;$A;$V;$P;$Q;$S;$EP;$EQ;$ES;$F;%periodic measurement%%"`
999998 - ID
36 - TEMP
0 - STAT
0 - DSTAT
300 - A
231 - V
134.0 - P
0.1 - Q
148.0 - S
707.0 - EP
0.5 - EQ
690.0 - ES
0.0 - F
published
`mosquito_pub -h soporte-tecnico.bitnamiapp.com -p 1883 -t LU/LUM/ACT/999998 -m "3;3;"`
received message in trace
after that published `1;` and DSTAT changed to 3, according to metadata `"SDIMM%3;$DSTAT;% set ligthing status%%"`

ralcaide, 2017-08-17:
DONE. Closing the issue

closed 2017-08-17 by ralcaide