Sinapse-Energia/AP-Sinapse #18: [ML 1 - Context - Design] To study and explain the options to save context variables in NVM

ralcaide, 2017-08-07:
The context variables should be saved in a Non Volatile Memory and the EPDs variables will be updated frequently. It is necessary to study the different options and choose the best one.

Problem: The writing in a flash sector requires the deletion of the whole sector and then write again the sector. We are working with 128 KB sectors, so we need 128 KB of "other memory" to save the previous sector during writing operations. We name this "other memory" as a swap memory and we were thinking to use RAM memory to do that but the RAM is very little, so it is not a good idea.

Possible solutions:

1. Use another sector of the Flash as a swap memory
2. To use EEPROM (the device has 512 KB) memory as a swap memory
3. To save all the Context variables in EEPROM instead of Flash

ralcaide, 2017-08-07:
related with #12 and #3

fpgamcu, 2017-08-07:
1. Use another sector of the Flash as a swap memory
Short algorithm of write:
erase swap - copy context sector to swap (changing needed data) - erase context sector - copy from swap to context
Timings:
2 sec max for erase swap + 3.3sec max to copy to swap + 2 sec to erase context + 3.3 sec to copy to context = 10.6 seconds
Using RTOS we can perform "erase swap" in background, but as I remember - CPU halts when writing to internal Flash is in progress
Another options is to use so called Paging system. Use both sectors and mark in each sector, that indicates what sector contains last data. When need to modify: erase oldest sector - write data changing it from newest to oldest sector and place freshen mark making this sector newer
Timings:
2 sec max for erase oldest + 3.3sec max to copy data = 5.3 sec
2. To use EEPROM (the device has 512 KB) memory as a swap memory
Short algorithm of write: 
copy context sector to swap (changing needed data) - erase context sector - copy from swap to context
Timings:
1.1sec to transmit to eeprom + 5.12 sec to write to eeprom + 1.1sec to receive from eeprom + 2 sec to erase context + 3.3 sec to copy to context = 12.6 sec (11.5 sec if receive from eeprom and write to flash at the same time)
3. To save all the Context variables in EEPROM instead of Flash
Short algorithm of write: 
just write what been changed
Timings:
10 ms both for write a single byte or 256 bytes page, 8 usec to transmit a byte, to read/write need to transmit 3 bytes of address and 1 byte of command, so 32 usec will be in any operation

fpgamcu, 2017-08-07:
Also MCU flash have 10K cycles of gгaranteed write
EEPROM have 1.2M cycles of gгaranteed write

ralcaide, 2017-08-07:
Thank you very much @fpgamcu.
So it seems that the best option is the third one, isn't?. What is your advice?

Could please also to add an analysis about these same operations but using a SD card instead of EEPROM?

fpgamcu, 2017-08-08:
I think that the best variant now is third.
For SD card we have SDIO peripheral in our MCU, but some of its pins is taken by other GPIO.
Using SDIO gives us a possibility to perform operations faster, than using of SPI for SD card.
Also we have FATFS module in CubeMX, so we can generate an SD card access library using CubeMX and even have our EPDs represented by files (binary files or even text files with some simple syntax, we can use text files because with SD card we will have several GB of space for context)
For the timing it is usually less than one second for several KB operations. Great plus that SD card have internal controller, and we have not to worry about the pages and other Flash specific things.

ralcaide, 2017-08-14:
Done. By the moment we will work with External EPROM but we plan to add an SD card in the new versions of the HW

closed 2017-08-14 by ralcaide