Sinapse-Energia/AP-Sinapse #23: [ML 2 - Command 3;X;] To develop the command 3;X; - ONLY SIMULATION - NO RF

ralcaide, 2017-08-14:
To implement all the operations related with the command 3; . Resume, see wiki: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#on-demmand

Steps for the desired behavior:

1. The AP receive a command 3;XX; through INSTALLATION_NAME/AP_ID/ACT/EPD_ID where
- INSTALLATION_NAME is in the Context (MQTT_Topic_Root)
- AP_ID is in the Context (Device_ID)
- ACT is fix
- EPD_ID is the ID of the EPD and should be in the Context. If the EPD_ID is not in the Context, the command should be ignored. Normally, if the EPD_ID is not in the Context, the AP is not subscribed to the above topic.
The received command is saved in the MQTT_RECV_queue
2. The message is processed when RTOS give time to this task. The process message at this stage is only the "Simulation" of the RF actuation. The simulation performs an update of the required EPD Context variables. 

Example: If the EPD 500001 was OFF (EPD::A = 0, EPD::V = 220 and EPD::P = 60) and arrives the message "3;100;" for this device. Then, the Context variables of this EPD should be: EPD::A = 272, EPD::V  = 220 and EPD::P = 60, because A unit is mA and V unit is Volts

This is a previous step of the real command implementation. The real command implementation will not update the Context, it will use the RF southbound functions in order to send a RF command

ralcaide, 2017-08-24:
TESTED with the following results:

- Simulation of the command: OK but very poor, to be improved in other issues
- Management of the limit cases: KO, the numbers > 100 are not truncated. To be fixed in other issues
- Work with Context : KO (but problem is also related with the command 1;). If we act over a device X and then we ask about other one like Y, the result of the DSTAT set in the X. To be fixed through other issue

fpgamcu, 2017-08-24:
@ralcaide About work with context Engine, there seems to be a bug with engine: when I set the EPD ID using next code `SetVariableThreadSafe("EPD::EPD", epdid);`, Engine not read the EPD data from NVM - not call GenericRead function.

fpgamcu, 2017-08-24:
> If we act over a device X and then we ask about other one like Y, the result of the DSTAT set in the X. To be fixed through other issue

Fixed this by adding `int	SwitchAndReadEPD (const char *id);` to NBinterface

ralcaide, 2017-08-25:
Yes, the Engine has a "bug" in this part.  The EPD is only swapped during the process of a frame. The swap is not automatically done setting this EPD::EPD. 
Anyway, we will integrate today a new version of the engine that will work as you expect. The swap-in will be done only setting the var EPD::EPD, then you could refactor your solution

ralcaide, 2017-09-18:
TESTED KO: The management of commands with out of the limits values are not well done. See #34. Waiting to this bug fix in order to close this issue

closed 2018-02-08 by ralcaide