Sinapse-Energia/AP-Sinapse #28: [ML 2 - API 314;] To develop the command 314; that erase the NVM

ralcaide, 2017-08-16:
To implement the command 314;

`314;password;`

Where password by the moment is "12345678".

This command removes the NVM and set in it the default Context values.

fpgamcu, 2017-08-16:
Now erase takes some time. About 15 sec. Should I add answer, for example `314;OK;`, when the command is finish erased NVM?

ralcaide, 2017-08-16:
Let me think about the response. I will create another issue if it is necessary

ralcaide, 2017-08-17:
After a little test, this command is not working as expected because after perform the 314; the command 313; is still sending a 713; with the maximum of nodes defined in context.h. All the nodes are the same.

Example: 
`713;200001;100;200001;100;200001;100;200001;100;200001;100;;`

After a 314; the 313; should not return nothing. There is zero EPD in the NVM

@fpgamcu , please fix it.

fpgamcu, 2017-08-17:
314; erases NVM, but there is still EPD data in the ram, and Context engine have no mechanism to remove EPD from ram. If you execute reboot command after 314; you will see, that 313; return an empty `713;`
Shall I execute reboot after 314; ?

ralcaide, 2017-08-17:
OK, I understand, I see already the 713; empty. 
No, by the moment we leave as it is. I move it again to "Testing"

ralcaide, 2017-08-23:
Tested OK. Closing the issue

closed 2017-08-23 by ralcaide