Sinapse-Energia/AP-Sinapse #30: [EPD Discovery 316 - 313 - 713] Deafness / HF after set up more than 20 EPDs

ralcaide, 2017-08-18:
The following MQTT.fx Script has been used in order to perform a charge test:

`var Thread = Java.type("java.lang.Thread");

function execute(action) {
    out("Test Script: " + action.getName());
    
    var topic_root = "LU/LUM";
    var id_ap = "700099";

    mqttManager.subscribe("LU/LUM/SEN");
    //ask 313;
    send_313(id_ap, topic_root);
    sleep_mqtt();
    send_316(id_ap, topic_root, "200001", "60");
    sleep_mqtt();
    send_313(id_ap, topic_root);
    sleep_mqtt();
    send_316(id_ap, topic_root, "200002", "60");
    sleep_mqtt();
    send_313(id_ap, topic_root);
    sleep_mqtt();
    send_316(id_ap, topic_root, "200003", "100");
    sleep_mqtt();
    send_313(id_ap, topic_root);
    sleep_mqtt();

    // Modifying nodes
    send_316(id_ap, topic_root, "200001", "90");
    sleep_mqtt();
    send_313(id_ap, topic_root);
    sleep_mqtt();
    send_316(id_ap, topic_root, "200002", "120");
    sleep_mqtt();
    send_313(id_ap, topic_root);
    sleep_mqtt();

    execute_charge_test(id_ap, topic_root);
    

    ///
    action.setExitCode(0);
    action.setResultText("done.");
    out("Test Script: Done");
    return action;
}

function send_313(id_ap, topic_root) {
  out("send_313");
  var topic = topic_root + "/ACT/" + id_ap;
  var msg = "313;";
  mqttManager.publish(topic, msg);

}

function send_316(id_ap, topic_root, id_epd, power) {
  out("send_316 - ID: " + id_epd + " P: " + power);
  var topic = topic_root + "/ACT/" + id_ap;
  var msg = "316;"+id_epd+";"+power+";";
  mqttManager.publish(topic, msg);  
}

function send_398(id_ap, topic_root, minutes) {
  out("send_398");
  var topic = topic_root + "/ACT/" + id_ap;
  var msg = "398;"+minutes+";";
  mqttManager.publish(topic, msg);  
}

function execute_charge_test(id_ap, topic_root) {
  epd_max_number = 40;
  start_epd = 200010;
  start_p = 60;
  for (i = 1; i <= epd_max_number; i++) {
    epd_id = start_epd + i;
    p = start_p + i; 
    
    send_316(id_ap, topic_root, epd_id.toString(), p.toString());
    sleep_mqtt();
    
    if (i % 10 === 0) { //Ask only each 10 times
      send_313(id_ap, topic_root);
      sleep_mqtt();
    }
  }
}

function out(message){
     output.print(message);
}

function sleep_mqtt() {
   Thread.sleep(2000);
}
`

The context.h has been modified in order to allow a maximum of 50 elements:

`#define		MAX_TABLE_SIZE		50`

During the execution of the script everything work OK until the i of execute_charge_test is > 20, the third time that the 313; is asked in this script doesn't return any answer. After that the device is not responding anymore. 
After a reset + run in Debug, a new 313; is sent and a HF is produced.

ralcaide, 2017-08-18:
Test performed again with similar results. Last messages listen by the AP:

Node 700099 <18-8-2017 15:43:21> Incoming message #316;200040;90;#

Node 700099 <18-8-2017 15:43:21> Incoming message #313;#

Then HF and the Device is not working anymore

fpgamcu, 2017-08-18:
The bug is in the MQTTAPPI.c
Line 16 : `#define BUFSIZE	256`
When we have many devices for `313;` command MQTT buffer overflows

ralcaide, 2017-08-19:
Thank you, I understand the problem but I'm not sure about what is the best solution. Possibilities:

1. Increase the size of the buffer: This is not a good solution because the 90% of our commands are smaller than 256 KB and and we will use more RAM than necessary
2. If the message is bigger than BUFSIZE, then split it in several messages smaller than BUFSIZE: This is a more complex solution but sustainable.

Anyway, before send a message, our FW should check the size in order to avoid HF. Also, if it is not possible to send a message maybe we should send an ERROR message (new API message to be defined)

@fpgamcu , what do you think about the possible solutions?, do you have any other idea?

ralcaide, 2017-08-23:
Tested OK. It works great

closed 2017-08-23 by ralcaide