Sinapse-Energia/AP-Sinapse #31: [ML 3 - Command 4;] To implement command 4; ONLY CONTEXT - NO RF

ralcaide, 2017-08-22:
To implement all the operations related with the command 4; . Resume. See Wiki:
https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#ask-status-periodically

It is similar to command 1; (Issue #22 ) but with scheduled behavior. 
1. The AP receive a command through INSTALLATION_NAME/AP_ID/ACT/EPD_ID where
2. The message is processed when RTOS give time to this task, the process of this message involves the following:

- Update the context of the EPD_ID with the periodic info received in the command (TO DEFINE in other issue)
- Creation of the needed scheduled tasks.The scheduled task involve the creation of the messages with the Context of the EPD_ID and save it in the MQTT_SEND_queue

3. The messages will be published after be saved in the MQTT_SEND_queue, when the FreeRTOS give time
See diagram flow: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Flow-diagrams#global-flow-diagram

TODO in other issue: in the scheduled task it is necessary to call the RF functions. The RF functions after receive a response will update the Context and in the next scheduled task the answer will be created with the updated Context.

fpgamcu, 2017-08-23:
What is the meaning of N parameter in command?

ralcaide, 2017-08-23:
You can download the whole API from here: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#summary-of-the-sinapse-api

Extract from this document:

Parameters: 

MIN:minutes between periodic publications.
H1H1:M1M1: Initial time for interval keeping that status. This value can be a Sunset / Sunrise based Offset (Important note 2)
H2H2:M2M2: Final time for interval keeping that status. his value can be a Sunset / Sunrise based Offset (Important note 2)
NTIMES: Indicates number of times that this order will be repeated
1….N: 1 time… N times
-1: Always

Example:
4;5;03:00;17:30;-1;
It is indicated to publish measurement in valid topic publication each 5 minutes from 3:00 AM to 17:30 PM always every day

fpgamcu, 2017-08-30:
Can we use ; instead of : in time format for commands 2; and 4;?
For example `4;3;02;00;17;00;25;`

ralcaide, 2017-08-30:
Normally not, we use ; in order to delimit parameters and the hour:min is only one parameter. 
Do you have problems with that? Is an Engine limitation? Let me know if it is an Engine limitation in order to manage this situation

fpgamcu, 2017-08-30:
We have hour and minutes in context as separate parameters, to set them using the engine we can only use ; separator.
But with new feature of parameters read/write hooks I can create hook to read HH:MM, parse and save to hours and minutes parameter.
Should I use such hook?

ralcaide, 2017-08-30:
Yes please, Thank you

ralcaide, 2017-09-05:
**Testing performed**

** Test 1**

SETUP 200001
then
"4;1;01:00;23:59;-1;" into "TESTING_AP/700099/ACT/200001"

Results
- After receive the command 4; the AP produces a HF
- After restart publish periodically (T = 1 min + several seconds) a message but these messages are KO:

1. 100;700099;200001;0;0;0;0.0;220.0;70.0;0.0;0;0.0;0;0.0;0;0;
2. 999;1;Unknown message;
3. 999;1;Unknown message;
4. 999;1;Unknown message;
....
N. 999;1;Unknown message;

ralcaide, 2017-09-12:
After several tests it seems almost OK but I see a little problem. The message is published in "MEASUREMENTS" instead of "PERIODIC"

`017-09-12 17:46:33 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;200001;23;1;50;158.5;220.5;35.0;0.0;35;0.0;0;0.0;48;1200;`

1; should publish in MEASUREMENTS
4; should publish in PERIODIC

ralcaide, 2017-09-12:
TESTED almost OK:

Topic: KO
Period: OK
Start time: OK
End time: OK
N times: OK

To close after test the fix of the topic

ralcaide, 2017-09-13:
TESTED TOPIC OK. Closing the issue

closed 2017-09-13 by ralcaide