Sinapse-Energia/AP-Sinapse #32: [ML 3 - Scheduled behavior in Context - EPD]  To design / implement how can be saved the scheduled behavior in the Context

ralcaide, 2017-08-22:
The EPDs (and APs) will have some kind of scheduled behavior related with the commands 2; and 4; . In order to implement these commands in a good way it is necessary to save in the Context the info related with the Schedules. 
At the beginning, the EPDs don't have any scheduled behavior and work as "on demmand" with the commands 1; and 3; but they can receive commands 2; and 4; in order to work ALSO with a schedule. If the AP is restarted or disconnected from Internet, the scheduled behavior should still works as expected. 

What should we save in the Context?

- 4; : Per EPD, it is necessary to save the period, the start hour, the end hour and the N present in the command 4;. Then the AP will publish the status of each EPD periodically depending the info saved in the Context. If the period is not set or <= 0, then, the status is not published periodically
- 2; : Per EPD, it is necessary to save its scheduled behavior. This schedule covers 24 Hours and have a maximum of 10 steps with a status (0..100). In the Context should be saved in some form these 10 steps (HH:MM) and status per EPD. If the EPD has a scheduled behavior set, the AP will act over the devices as its schedule says. 

What to do in this issue:
- Design & Define how the necessary data should be saved in the Context in order to need the less possible memory. Codification?
- To modify the Context in order to have this info about the scheduled behavior.

fpgamcu, 2017-08-23:
Necessary parameters that will be saved in the context for command 4;
EPD_Ask_status_periodically_period_minutes EPD::ASPPM 1 byte integer
EPD_Ask_status_periodically_start_hour EPD::ASPSH 1 byte integer
EPD_Ask_status_periodically_start_minute EPD::ASPSM 1 byte integer
EPD_Ask_status_periodically_end_hour EPD::ASPEH 1 byte integer
EPD_Ask_status_periodically_end_minute EPD::ASPEM 1 byte integer
EPD_Ask_status_periodically_N EPD::ASPN 1 byte integer (not clearly meaning of N for me, maybe need more bytes?)
6 bytes total for each EPD

Necessary parameters that will be saved in the context for command 2;
EPD_Lightning_profile_dim_1 EPD::LPD1 1 byte integer
EPD_Lightning_profile_hour_1 EPD::LPH1 1 byte integer
EPD_Lightning_profile_minute_1 EPD::LPM1 1 byte integer
EPD_Lightning_profile_dim_2 EPD::LPD2 1 byte integer
EPD_Lightning_profile_hour_2 EPD::LPH2 1 byte integer
EPD_Lightning_profile_minute_2 EPD::LPM2 1 byte integer
EPD_Lightning_profile_dim_3 EPD::LPD3 1 byte integer
EPD_Lightning_profile_hour_3 EPD::LPH3 1 byte integer
EPD_Lightning_profile_minute_3 EPD::LPM3 1 byte integer
EPD_Lightning_profile_dim_4 EPD::LPD4 1 byte integer
EPD_Lightning_profile_hour_4 EPD::LPH4 1 byte integer
EPD_Lightning_profile_minute_4 EPD::LPM4 1 byte integer
EPD_Lightning_profile_dim_5 EPD::LPD5 1 byte integer
EPD_Lightning_profile_hour_5 EPD::LPH5 1 byte integer
EPD_Lightning_profile_minute_5 EPD::LPM5 1 byte integer
EPD_Lightning_profile_dim_6 EPD::LPD6 1 byte integer
EPD_Lightning_profile_hour_6 EPD::LPH6 1 byte integer
EPD_Lightning_profile_minute_6 EPD::LPM6 1 byte integer
EPD_Lightning_profile_dim_7 EPD::LPD7 1 byte integer
EPD_Lightning_profile_hour_7 EPD::LPH7 1 byte integer
EPD_Lightning_profile_minute_7 EPD::LPM7 1 byte integer
EPD_Lightning_profile_dim_8 EPD::LPD8 1 byte integer
EPD_Lightning_profile_hour_8 EPD::LPH8 1 byte integer
EPD_Lightning_profile_minute_8 EPD::LPM8 1 byte integer
EPD_Lightning_profile_dim_9 EPD::LPD9 1 byte integer
EPD_Lightning_profile_hour_9 EPD::LPH9 1 byte integer
EPD_Lightning_profile_minute_9 EPD::LPM9 1 byte integer
EPD_Lightning_profile_dim_10 EPD::LPD10 1 byte integer
EPD_Lightning_profile_hour_10 EPD::LPH10 1 byte integer
EPD_Lightning_profile_minute_10 EPD::LPM10 1 byte integer
30 bytes total for each EPD

For each EPD we need to add 36 bytes, we have 13 bytes reserved now, so EPD context size will be increased from 208 bytes to 231 bytes
for 208 size EPD context we can store 627 EPDs in 128K of memory

for more efficient of usage the memory we can change device context size from 656 to 647 (reduce reserved from 66 to 57 bytes) and context size from 231 to 235 (add 4 bytes reserve)
this will allow us to store 555 EPDs in 128K memory

now we have 256K memory - we can store 1112 EPDs with EPD size 235 bytes

ralcaide, 2017-08-23:
Thanks for the design.
I was thinking about to use less memory to do that, for example in 2; to code the DIM + hour:minute in 2 bytes:

- 24 hours = 1440 minutes  - > 11 bits
- 5 bits to codify the DIM

@fpgamcu , what do you think about that? I'm not sure if its worth to implement that

ralcaide, 2017-08-23:
By the moment we will implement the solution proposed by @fpgamcu but reserving already 50 bytes more per EPD:
- 36 bytes : Proposed solution
- 14 bytes : Reserved

closed 2018-02-08 by ralcaide