Sinapse-Energia/AP-Sinapse #33: [ML 3 - Command 2;] To implement command 2; ONLY CONTEXT - NO RF

ralcaide, 2017-08-22:
To implement all the operations related with the command 2; . Resume. See Wiki:
https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#lighting-profile

It is similar to command 3; (Issue #22 ) but with scheduled behavior.

1. The AP receive a command through INSTALLATION_NAME/AP_ID/ACT/EPD_ID where
2. The message is processed when RTOS give time to this task, the process of this message involves the following:
- Update the context of the EPD_ID with the scheduled received in the command (To Be Defined in #32 )
- Creation of the needed scheduled tasks.The scheduled tasks involve the update of the Context with the scheduled stat (SIMULATION). With RF SB, here should be called the RF functions in order to act over the EPDs  

See diagram flow: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Flow-diagrams#global-flow-diagram
TODO in other issue: in the scheduled task it is necessary to call the RF functions. The RF functions will act over the EPD and after receive a response will update the Context. The scheduled task ends calling the RF SB in order to act over the EPDs

fpgamcu, 2017-08-29:
Have this command variable length, or it always have all 30 parameters?
Should there be a way to cancel scheduled actions, for example set DIM to -1 (2;-1;00;00;-1;00;00;......) ?

ralcaide, 2017-08-29:
- Length: Is variable with a maximum of ten steps. Is this a problem?
- To cancel a schedule of an EPD:  2;-1;00:00; . This is not explained in the API but will be updated

ralcaide, 2017-08-29:
See: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#lighting-profile

fpgamcu, 2017-08-29:
No, dynamic length is not a problem, but command will be processed outside the engine. As I know engine is not support dynamic command length (except list of EPDs)?
2;-1;00:00; cancel all profile for an EPD, if I understand right.

ralcaide, 2017-08-29:
You are right, at this moment the engine does not support the dynamic command length but I think is a good idea to process  it inside of the engine. Please consider that the length will be fix with the following examples:

- 1 step: 2;100;01:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;
- 2 steps: 2;100;01:00;0;05:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;
- Delete: 2;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;-1;00:00;

Basically, when a step start with -1 should be ignored

This allow the implementation inside of the engine?

fpgamcu, 2017-08-29:
I think yes. Now will check.

ralcaide, 2017-09-12:
TESTED OK (With ETH and time starting at 00:00)

Steps: OK
Delete scheduler: OK

ralcaide, 2017-09-18:
TESTED with RTC OK

closed 2017-09-18 by ralcaide