Sinapse-Energia/AP-Sinapse #34: [ML 2 - Improvements command 1 & 3] Bugs found and to be fixed during the test of commands 1 & 3 - ONLY CONTEXT

ralcaide, 2017-08-24:
During the test have been found the following behavior:

- Act over 200001 with 3;100;
- Ask over 200001 with 1;, result OK : 100;700099;200001;0;0;100;0;0;100;0;0;0;0;0;0;0;
- Act over 200002 with 3;50;
- Ask over 200002 with 1; result OK: 100;700099;200002;0;0;50;0;0;100;0;0;0;0;0;0;0;
- Ask over 200001 with 1; result **KO**: 100;700099;200001;0;0;**50**;0;0;100;0;0;0;0;0;0;0; . Expected DSTAT = 100. The previous actuation was over 200002
- Act over 200001 with 3;150; 
- Ask over 200001 with 1; result **KO**: 100;700099;200001;0;0;**150**;0;0;100;0;0;0;0;0;0;0;. Expected DSTAT = 100. 
- Act over 200001 with 3;-50; 
- Ask over 200001 with 1; result **KO**: 100;700099;200001;0;0;**-50**;0;0;100;0;0;0;0;0;0;0;. Expected DSTAT = 100. 

Bugs to be fixed:

- The actuation over one EPD should not modify the values of other EPDs
- The FW should truncate the not valid values. DSTAT should be  0 <= DSTAT <= 100 . If the AP receive a value < 0 should set 0 and of receive a value > 100 should limit it to 100

fpgamcu, 2017-08-30:
I found another bug related with these one. Now for check the borders of value I use mechanism introduced in new engine (file wrsouth.cpp line 110) 
I use next check `if (dim < 0 || dim > 100) return NULL;` but when I return NULL instead of value, engine still set the parameter with wrong value, 101 or 150 for examlpe.

ralcaide, 2017-09-05:
After WIFI integration is still appearing the hard fault. See attached file

[milestones_1_2_050917_0003_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1279116/milestones_1_2_050917_0003_API_ENGINE_WIFI_700099.txt)

ralcaide, 2017-09-12:
TEST: In progress.

TESTS have been performed with good results. To test the limits regarding @fpgamcu comments

ralcaide, 2017-09-13:
@fpgamcu , regarding your previous comment:

It is necessary to modify the function *Write_DSTAT in order to fit our requirements. This is not an engine function, is a function called from the engine (hook). The actual code (with relays relays actuation included) is designed to be executed in an EPD, not in an AP. The AP should have other function to write the Write_DSTAT because the AP should not manage the relays, neither ignore the values bigger than 100 or lower than 0. If DSTAT > 100  should be limited to 100 and if DSTAT is negative, should be set to 0.
Could you please modify this function in order to fit the explained requirements?

fpgamcu, 2017-09-13:
Sorry, from comment I am not understand well what need to be done.
1) Add code to set dim to 0 if <0 and to set to 100 if >100
Any other changes?

ralcaide, 2017-09-13:
yes, in the simulation, only that.

Then, in the RF, it will be something like that. If the user send a 3;150; the RF should send the dimming to 100 instead of 150

ralcaide, 2017-09-18:
Tested KO. Waiting for the fix in order to closed it

fpgamcu, 2017-09-19:
What is the proper way to check variable value borders using the engine?

In set RFFL command I used next code in hook (simplified code) `if (value != "0" && value != "1") return NULL` to prevent setting not allowed value, but when I pass '2' with command, this value is still set in the context.

In DSTAT hook i used another way to limit the value (simplified code) `if (value < "0") return "0"` but it also don't limit value.

Should I check value and limit it outside the engine?

ralcaide, 2017-09-19:
let me check it with the Engine Team

fpgamcu, 2017-09-19:
ok, thank you)

fpgamcu, 2017-10-25:
I added code to limit received DIM into Process_RF_RECV_queue function. And added code to prevent changing one EPD by other. Still, if we will receive response with non existing EPD ID we will corrupt data of last used EPD. Is there in engine function like bool isEPDexist(const char* EPDID). It will help a lot with checking EPD ID of received packet.

closed 2018-02-08 by ralcaide