Sinapse-Energia/AP-Sinapse #35: [ML 3 - Simulation of 3; 1; & 100; - Improvement] To improve the simulation of the commands 3;, 1; and 100;

ralcaide, 2017-08-24:
Issues related with #22 & #23 
The AP should publish ONLY valid messages and currently the simulated messages are NOT valid and very poor. 
The AP should have two modes of behavior decided during the compilation time: RF or SIMULATION
If SIMULATION is selected (this issue is related with this mode), the simulated messages 100; should have correct values.

For example, if we have a device 200001 with P = 100W (created with 316;200001;100; in the AP 700099), the following behavior is expected: 

- 1; (after the setup) : 100;700099;200001;0;0;0;220.0;0;0;0;0;0;0;0;0;0;
- 3;100;
- 1 : 100;700099;200001;30;1;100;454.4;220.0;100.0;0.0;100;0.0;0;0.0;50;1503583747;
After 1 minute:
- 3; 50;
- 1; 100;700099;200001;25;1;50;227.3;220.0;50.0;0.0;50;1.6;0;1.6;50;1503583889;
- After 1 hour:
- 3;0;
- 1; 100;700099;200001;20;0;0;0.0;220.0;0.0;0.0;0;50.1;0;50.1;50;1503584180;

In the Sinapse API it is possible to see the meaning and units of each value. Normally, these values will be set by the RF part but the simulator should generate also valid values. Following a little explanation about how should work each value in the simulator:

ID: 100;
AP_ID: 700099;
EPD_ID: 200002;
T: 30; This temperature depend of the DSTAT, for example: 20 + (DSTAT/10) + rand(-2..2)
STAT: 0; : Depends of the DSTAT. If DSTAT is > 0, then 1. If it is 0, then 0
DSTAT: 10; : Set by the command 3;
A: 0.0; : Depends of P . I = (P / V) * 1000, the unit is mA
V: 220.0; : 220 + rand (-5..5)
P: 100.0; : Depends of nominal power and DSTAT. P = PNom * (DSTAT / 100 )
Q: 0.0; : always 0 because we suppose that cos phi = 1
S: 0; = P but integer because cos phi = 1
EP: 1.6; Energy consumed since set up in Wh
EQ: 0; Always 0
ES: 1.0; Like EP
F: 50; Frequency in Hz, for example: 50 + rand(-2..2)
Timestamp = Valid timestamp that represents the last update of the V. The V should be updated each time that a command 3; arrives

At this moment all the messages published by the AP are not valid because for example DSTAT is > 0 and STAT = 1 

It is necessary to create a realistic simulator

fpgamcu, 2017-08-29:
Should I calculate EP in the Context daemon?

fpgamcu, 2017-08-29:
Sorry, wrong button :(

ralcaide, 2017-08-29:
Yes, please

ralcaide, 2017-09-12:
TESTED OK

The time-stamp is not good because the time is not got from NTP. To close after test with real time

ralcaide, 2017-09-18:
TESTED OK after set the RTC

closed 2018-02-08 by ralcaide