Sinapse-Energia/AP-Sinapse #36: [Engine Integration] To integrate a new version of the API engine

ralcaide, 2017-08-25:
The new version will contains the following improvements:

- Flush method: Save the current context into the NVM (swap-out)
- Swap: The swap-in will be performed when the EPD::EPD var is set and not only during the frame process
- All the hooks to the functions to be executed during a a frame process or var process is in parametrizacion.cpp

The only file to me modified in order to add features to the API Engine is the parametrizacion.cpp

ralcaide, 2017-08-25:
@fpgamcu new version of the API engine has been integrated and uploaded. Please test that everything works as expected

fpgamcu, 2017-08-29:
Please describe mechanism (or maybe there is function) for creating new EPD with reading its parameters from nvm (now when executing `SetVariable("EPD::EPD", epdid);` for new ID engine creates EPD and writes its context to NVM. It is ok, but for reading EPDs on startup we need also a function that will create EPD and read from NVM first).

Now I implemented these feature with temporally disable write to NVM on startup - to disallow engine write to NVM when creating new EPDs on startup. But maybe it is not a good way.

Also there is a bug in engine: when adding new EPD with 316; when no EPD exists before (after clear NVM with 314;, 313; return no EPD) engine creates two EPDs, one with ID specified in 316; and another with empty id. And these empty ID EPD writes to the NVM with index 0. Because these bug empty ID EPD also added when loading EPDs on startup.

ralcaide, 2017-08-29:
I don't understand very well the first problem. Why do you need to create EPDs on startup? all the EPDs are in NVM and should be read from there. I suppose that the problem is related with the population of the array that contains the EPDs, on startup this array is empty and is not automatically populated, isn't? I suppose also you are populating this array on startup reading from NVM but disabling the writing into the NVM in order to avoid double EPD creation in the NVM. Is it like that?

In relation to the second problem, only appears when the list of EPDs is empty? We will research about it

fpgamcu, 2017-08-29:
About first: you are right :) under creation i mean the population of the array from NVM. Is the way I described "disabling the writing" is ok?
About second: yes, only appears when the list of EPDs is empty

ralcaide, 2017-08-29:
1. OK, by the moment leave it like that. I will think if there is a better solution
2. OK, I will explain this bug to the API Engine team

ralcaide, 2017-08-31:
@fpgamcu , could you apply the attached patches to your code, test that everything works OK and then commit the code?. These patches:

- Fix the problem with the double creation of EPDs when the NVM is empty
- Added the $$ as a sigil in order to create EPDs. Now the command 316; should have $$EPD_ID::EPD_ID instead of $EPD_ID::EPD_ID . Explanation:

_The proposal is to "bless" the $ sigil with an optional extension (for instance a second $, a ! or any other non alphabetic ASCII char...), as a signal of privileged mark allowing the creation of records.
For example, if the EPD subspace primary key (EPD::EPD) apear "as is" ($EPD::EPD) in a command, it means that is NOT allowed to create a record is the value doesnt exist, but if it appears blessed ("$$EPD::EPD"), it means that it is ALLOWED to create a new record is no one does exist for that value._

[0014-JRB-33-Handling-sigile-blessing-in-order-to-allow-cr.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1267046/0014-JRB-33-Handling-sigile-blessing-in-order-to-allow-cr.patch.txt)
[0015-JRB-33-Commands-and-variables-to-test-sigil-blessing.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1267047/0015-JRB-33-Commands-and-variables-to-test-sigil-blessing.patch.txt)
[0016-JRB-Some-cleaning-comments-and-description-spelling.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1267048/0016-JRB-Some-cleaning-comments-and-description-spelling.patch.txt)

fpgamcu, 2017-08-31:
Yes, of course. I will integrate it now.

ralcaide, 2017-09-01:
New patches for the API Engine, this version populate the array of EPDs automatically after the startup. 

Explanation

>  There are now a couple of new methods for restoring the NVM to the context. (a instance method, and a static one to call the former in every existing subspace).
The initial idea of driving it from the class constructor doesn`t work , because at this time, the Subspace is not yet populated, and there has no idea of which elements it will have to hold.
So (and this is very important) the Restore() method has to be called after all the Context is created and fully populated. In order to avoid to call CreateContext() and CSubspace::Restore() in sequence, there is now a call to Restore, at the very last line in the CreateContext function.
Has to be in this way. Is not possible to trigger the Restore() automatically

@fpgamcu, could you integrate them?

[0017-JRB-34-WriteData-function-factored-out-from-SwapIn.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1269609/0017-JRB-34-WriteData-function-factored-out-from-SwapIn.patch.txt)
[0018-JRB-34-Changes-in-GenericReadElement-return-value-in.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1269610/0018-JRB-34-Changes-in-GenericReadElement-return-value-in.patch.txt)
[0019-JRB-34-New-methosds-for-restoring-the-NVM-in-re-star.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1269611/0019-JRB-34-New-methosds-for-restoring-the-NVM-in-re-star.patch.txt)

ralcaide, 2017-09-04:
We've found the problem with the empty EPD. We will implement a patch to fix that but in the meanwhile it is possible to avoid the problem removing the null string from the EPD::EPD entry (line 348 of the parametrization.cpp file 
`new CVariable ("EPD_ID", nvmEPD[0].nick, new CVarstring("") )`

This null string is not necessary and is creating the problem

fpgamcu, 2017-09-05:
Also I noticed that I must not change data returned by GetVariable() function, because for string parameters this action will change the parameter in memory. Maybe GetVariable() must return pointer not to parameter buffer but to some independent buffer in memory, to protect parameter from changing.

ralcaide, 2017-09-05:
Why should return a copy instead of the real value? I think you get the variable to read it or to modify it, isn't?

fpgamcu, 2017-09-08:
It depends on GetVariable() and SetVariable() functions contract. If for example we will have some check in SetVariable() (URI check etc) with access using GetVariable() returning pointer we can change URL without check and this can be security issue.

closed 2018-02-08 by ralcaide