Sinapse-Energia/AP-Sinapse #37: [ML 4 - Command 5;] To develop the command 5; related with the change of the MQTT broker

ralcaide, 2017-08-29:
To implement the command 5; explained here: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#change-mqtt-broker-credentials

The context contains a main broker and a backup broker. At the startup, both are the same but it should be possible to change these brokers (at least the main) in runtime.

The behaviour should be the following one:

The point is to have TWO sets of broker parameters: ONE, let´s call it the "configured parameters" (MAIN), and the OTHER, that we can call "last successful connection" (BACKUP).
The first one comes initially from the config, and changes each time a command 5; is received.
The second one is automatically updated upon successful broker connection.
Both are stored in the Context and both deserve to persist.
The connection process, wherever, first takes the first ONE.
If succeds, updates the second with the used parameters.
If fails, keeps the first contents, but retries to connect wit the second one.

fpgamcu, 2017-08-30:
Can we change format to `5;URI:PORT;USR;PASS;` ?

ralcaide, 2017-08-30:
Normally not, is it a big problem to use the current format?. If it is much better we can think about that but we prefer not to change the format

fpgamcu, 2017-08-30:
now in context description we have server name as single parameter like host:port
but I think I can change the context and make this parameters as two separate.
Should I add an Port parameter?

ralcaide, 2017-08-30:
Ok, yes please. Add another parameter to the context

ralcaide, 2017-09-18:
This command is not working well using WIFI Module due to a problem with the HW module connection. It will be fixed after the modification of the HW.

GPRS: TO TEST

ralcaide, 2018-02-08:
To be tested

ralcaide, 2018-04-25:
To be tested through C81

soporteHW, 2018-10-23:
Last test performed in FW V1.40.1

Test result: [5; Change MQTT Broker credentials](https://sinapseenergia.testrail.io/index.php?/tests/view/1108)

closed 2018-10-23 by soporteHW