Sinapse-Energia/AP-Sinapse #38: [ML 4 - Command 120;] To implement the command 120; - Set Group ID - NO RF

ralcaide, 2017-08-29:
To implement the command 120; here explained: https://github.com/Sinapse-Energia/AP-Sinapse/wiki/Sinapse-API#set-group-id

This command involves a call to RF AND an update of the Context. In this issue will be updated only the Context, the RF will be performed in other issue. It is necessary to update EPD::GRPID in order to set it to the received value. 

The Group ID is a 6 HEX string and it could be from "000000" to "FFFFFF" . If the received value is not correct should be ignored

ralcaide, 2017-09-18:
Not possible to test it now, waiting to the implementation of the command 202; (To explain by RAE)

ralcaide, 2017-12-16:
This command is not working:

- Send 120;FFFF11; to an EPD (100927) and then ask for its configuration using 202; I'm receiving 602;000007;100927;2;2017/12/16,22:08:13;0000;-1;-1;6;0;. The group ID is 0000 instead FFFF11.
- Same problem if I use other ID for groups like 100001 or 500001

@fpgamcu , could you fix that, please?

ralcaide, 2017-12-16:
Also, in some cases don't appears the group ID. what is the meaning of that?

602;000004;FFFFFF;2;2017/12/16,22:45:46;-1;-1;6;0; -> Here is missed the group id . Why??

fpgamcu, 2017-12-17:
@ralcaide About EPD::GPRID - according to context table it have type HEX and 2 bytes, so you can write there only 4 hex digits, like FAB7. Should I change it to 3 bytes?
About missing something in response formatting by the context engine - all questions to the context engine.

ralcaide, 2017-12-17:
As is explained in the issue, the Group ID is composed by 6 HEX characters but actually the 4 first characters will be always "FFFF". Two options:
- Increase the size in the context
- Save only the last 2 characters and add automatically the "FFFF" at the beginning. Then the command 120 should contains only 2 characters: 120;FF;. The context should have only 1 byte in this case  

@fpgamcu, what is the best option for you?

This command provokes sending of RF command to set the Group ID in the EPD. The RF command contains only the last two group id characters. See the wiki:

> Set Group Address Command to set the Group Address of an EPD. It has one parameter and the response to this message is a generic ACK
> GR : Byte that contains the last byte of the group address and goes from 0x00 to 0xFF. The first bytes of a group address are always 0xFF 0xFF

The group address should work as a the broadcast address but will be not automatically in the Context. The user is able to create an EPD that represents a Group, for example: FFFF11. And the act over this EPD. 
**Acting over this EPD should send RF commands related with ECs and SEs devices.**

ralcaide, 2017-12-18:
Thank you, I will test this and will let you know if everything works as expected

ralcaide, 2018-02-08:
To be tested

ralcaide, 2018-04-25:
To be tested through C80

closed 2019-03-28 by ralcaide