Sinapse-Energia/AP-Sinapse #40: [BUG ML 1-2-3  RTOS Behavior] Commands missed or very late processed

ralcaide, 2017-08-30:
A global test has been performed executing only commands 313 & 316 in order to verify the correct behavior of the Engine and RTOS but the result is not good. After several commands the AP stop processing messages or process them very late. Some messages 316; are lost or not processed. 

Attached the results of the test. It seems a RTOS problem or QUEUE problem. Do you have any idea @fpgamcu 

[milestones_1_2_290817_0001.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1263941/milestones_1_2_290817_0001.txt)

ralcaide, 2017-08-30:
Adding the results of a new test. Here you can see : 
- That the commands need more time and more time to be processed
- Also there are several commands (316;) missed or not processed
- Also the command 713; is not always correct, see the last two 713; it contains repeated EPDs

[milestones_1_2_290817_0002.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1264023/milestones_1_2_290817_0002.txt)

ralcaide, 2017-08-30:
More info:

After perform a 3;100; over 200001 and then 1; we got a HF. Then, restart and 313; and another HF

fpgamcu, 2017-08-30:
I think problem about delays and ignored incoming packets can be in the UART interrupt and circular buffer. For send feature I use my own buffer. But for receive i use existing circular buffer code. Now i working on change the receive buffer to safe one, I will finish implementing and testing tonight and push code in the morning (with new commands from 4-th issue too :) )

About duplicating you can notice that duplicate is always first element, that is element created because of bug. I will test it more tonight too.

Thanks for info. I will perform heavy testing on my side tonight to investigate HF problem.

ralcaide, 2017-08-30:
Before the introduction of RTOS I've tested with much better results. The losing of messages maybe are related with the circular buffer but the delays I don't think so, See the results of the second test, the answers are slowly and slowly, I think this is related with the RTOS

Regarding the duplication, I will integrate a new version of the Engine ASAP, normally this problem will be fixed

fpgamcu, 2017-08-30:
Can you please perform the test also using commit https://github.com/Sinapse-Energia/AP-Sinapse/commit/bbb9141ac59d8df521d91db1a063346e9bdfa2f5 to check the RTOS reaction time before new Engine version is integrated?

ralcaide, 2017-08-30:
Ok, I will do that tomorrow

ralcaide, 2017-08-31:
Test done with rev bbb9141. The reaction was great, as should be. Anyway the command 313 & 713 were not working good. See attached results

[milestones_1_2_310817_0003_Rev_bbb9141.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1266423/milestones_1_2_310817_0003_Rev_bbb9141.txt)

fpgamcu, 2017-08-31:
strange, i have different results 
[mqtt_31_08_17_12_02.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1266486/mqtt_31_08_17_12_02.txt)

ralcaide, 2017-08-31:
Results of the test performed with the @fpgamcu code/device

[milestones_1_2_310817_0004_SvaCode_999998.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1266812/milestones_1_2_310817_0004_SvaCode_999998.txt)

fpgamcu, 2017-08-31:
This commit https://github.com/Sinapse-Energia/AP-Sinapse/commit/a635e9ec205b6ec39e7a7c5b085908cde06784e3 
plus patch [0001-Correct-the-code-to-integrate-the-engine.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1266972/0001-Correct-the-code-to-integrate-the-engine.patch.txt)

ralcaide, 2017-08-31:
Testing with the version https://github.com/Sinapse-Energia/AP-Sinapse/commit/a635e9ec205b6ec39e7a7c5b085908cde06784e3 + the @fpgamcu patch and the results in relation to the  time reaction are OK but the 713; is wrong

[milestones_1_2_310817_0005_API_ENGINE_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1266982/milestones_1_2_310817_0005_API_ENGINE_700099.txt)

ralcaide, 2017-09-11:
Testing after including the needed time to process. Results
- Each GenericWriteElement needs more than 3 seconds
- Each command 316; needs more than 6 seconds (2 times GenericWriteElement, why @fpgamcu ?)
- Each time the scheduler is executed (1 minute) the scheduler needs more than NumberOfEPDS * 3 seconds. So, more than 20 EPDs need more than a minute

Possible solutions:

- Decrease the time to write an element
- Decrease the number of writtings

What do you think @fpgamcu ?

Example of results:

> 2017-09-11

 16:56:24 INFO TEST: STARTING AT: 2017-09-11 16:56:24 +0200
2017-09-11 16:56:24 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 313;
2017-09-11 16:56:24 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:40> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#
2017-09-11 16:56:25 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:40> ProcessMessage 0 msec
2017-09-11 16:56:25 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;
2017-09-11 16:56:30 INFO TEST: Executing basic setup
2017-09-11 16:56:30 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 316;200001;60;
2017-09-11 16:56:30 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> Incoming message #TESTING_AP/CMC/ACT/700099#:#316;200001;60;#
2017-09-11 16:56:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> Start saving Device context to NVM
2017-09-11 16:56:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> GenericWriteElement 301 msec
2017-09-11 16:56:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> ProcessMessage 319 msec
2017-09-11 16:56:36 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 313;
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> er GenericReadElement 0 msec
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:02:50> ProcessMessage 18 msec
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;200001;60;
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> Starting Scheduler action
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> GenericWriteElement 6 msec
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> er GenericReadElement 0 msec
2017-09-11 16:56:37 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> Ending Scheduler action
2017-09-11 16:56:38 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> SaveDevParamsToNVM 7220 msec
2017-09-11 16:56:38 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> End saving Device context to NVM
2017-09-11 16:56:42 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 316;200002;80;
2017-09-11 16:56:43 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> Incoming message #TESTING_AP/CMC/ACT/700099#:#316;200002;80;#
2017-09-11 16:56:46 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:00> GenericWriteElement 3050 msec
2017-09-11 16:56:48 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 313;
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> GenericWriteElement 3050 msec
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> ProcessMessage 6123 msec

TEST: Command 316; takes more than 6 seconds -> two times GenericWriteElement

2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> ok GenericReadElement 8 msec
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> ok GenericReadElement 8 msec
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> ProcessMessage 35 msec
2017-09-11 16:56:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;200001;60;200002;80;
2017-09-11 16:56:54 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 316;200003;100;
2017-09-11 16:56:54 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:10> Incoming message #TESTING_AP/CMC/ACT/700099#:#316;200003;100;#
2017-09-11 16:56:58 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> GenericWriteElement 3049 msec
2017-09-11 16:57:00 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 313;
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> GenericWriteElement 3050 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> ProcessMessage 6123 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> ok GenericReadElement 8 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> ok GenericReadElement 9 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> ok GenericReadElement 8 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:03:20> ProcessMessage 53 msec
2017-09-11 16:57:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;200001;60;200002;80;200003;100;
[... cut ...]
2017-09-11 16:59:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:05:50> ok GenericReadElement 8 msec
2017-09-11 16:59:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:05:50> ProcessMessage 249 msec
2017-09-11 16:59:31 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;200001;0;200002;90;200003;110;200100;60;200101;61;200102;62;200103;63;200104;64;200105;65;200106;66;200107;67;200108;68;200109;69;200110;70;
2017-09-11 16:59:36 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 316;200118;78;

ralcaide, 2017-09-18:
Tested OK

closed 2017-09-18 by ralcaide