Sinapse-Energia/AP-Sinapse #41: [ML 4 - 5] To integrate WIFI/ETH driver

ralcaide, 2017-09-01:
Instead of GPRS is also possible to use USR-232 driver (WIFI / ETH)

Manual is in the master
Attached socket code used in the bootloader
[Socket_bank.c.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1270353/Socket_bank.c.txt)
[Wifi232.c.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1270354/Wifi232.c.txt)
[Socket_bank.h.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1270357/Socket_bank.h.txt)
[Wifi232.h.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1270356/Wifi232.h.txt)

ralcaide, 2017-09-05:
First test using WIFI driver we are still loosing data and there is also a big delay in the process after several commands. See attached results 
[milestones_1_2_050917_0001_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1278929/milestones_1_2_050917_0001_API_ENGINE_WIFI_700099.txt)
file

ralcaide, 2017-09-05:
Deafness still appears with WIFI driver. The problem was not GPRS

[milestones_1_2_050917_0002_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1279096/milestones_1_2_050917_0002_API_ENGINE_WIFI_700099.txt)

ralcaide, 2017-09-07:
Testing the new version:

- The speed is still low
- The command 313; is returning a bad response, for example appears several EPDs repeated

See attached document:

[milestones_1_2_070917_0001_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1284789/milestones_1_2_070917_0001_API_ENGINE_WIFI_700099.txt)

ralcaide, 2017-09-11:
Tests performed with the new version of the code 4edda996beb9576d149f28b4307637fb49c9861c with the following global results:

- With only three EPDs the AP is working almost good but with little problems: The 313; is returning bad results sometimes and the process of 1; is taking more and more time. See attached results
[milestones_1_2_3_basic_setup_basic_actuation_080917_0001_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1291615/milestones_1_2_3_basic_setup_basic_actuation_080917_0001_API_ENGINE_WIFI_700099.txt)

- With more EPDs the AP is NOT working good. The reaction is very slow, the 313; is returning bad results always. See attached results
[milestones_1_2_3_basic_setup_advanced_setup_080917_0002_API_ENGINE_WIFI_700099.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1291620/milestones_1_2_3_basic_setup_advanced_setup_080917_0002_API_ENGINE_WIFI_700099.txt)

ralcaide, 2017-09-13:
TESTED with 100 EPDs separated 2 seconds : OK

ralcaide, 2017-09-13:
It will be tested also with 400 EPDs

fpgamcu, 2017-09-13:
I have found important bug with WiFi interaction. AT commands integrated in the driver works ok, but changing server host:port takes effect only after WiFi module reload (only possibly is power off on our board). The most important things is:
1) You can change server/port using 5; command, new server/port will be written to the WiFi module, but module will connect to old server until WiFi module reboot.
2) Because of these bug backup server feature is not work correct.
Way to fix: solder WiFi reboot pin with wire to some MCU pin.
With GPRS driver all must work ok.

ralcaide, 2017-09-13:
Thank you very much for the info. We will wire the WiFi reboot pin. By the moment, we leave like that.

One question: Rebooting the device with 398; does not reboot the WiFi module?

fpgamcu, 2017-09-13:
yes, 398; does not reboot the WiFi module

ralcaide, 2018-02-08:
Done and tested

closed 2018-02-08 by ralcaide