Sinapse-Energia/AP-Sinapse #42: [ML 4 - 5 ] Actual time is not set when working with WIFI/ETH

ralcaide, 2017-09-12:
After each restart of the device, the time is always 00:00 when working with ETH. Please get it from NTP after each restart (Like in GPRS)

Example:
`2017-09-12 12:29:52 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <0-0-0 00:01:20> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#`

ralcaide, 2017-09-15:
Due to a problem with the WiFi hardware is not possible to get the time from NTP Server. It will be implemented a new API command to set the actual time: 

`127;yyyy;mm;dd;hh;mm;ss;`

ralcaide, 2017-09-15:
Testing with 127;2017;09;16;00;36;36; but then we got the messages with 2 hours more:

Node 700099 <16-9-2017 02:39:51> Sended Ping getting 1

Why?. The 127; contains UTC and the AP add an offset depending of where is placed?

fpgamcu, 2017-09-15:
There is variable in the utils.c file in line 14 `int utcoffset = 2;` that determine offset added to the time

ralcaide, 2017-09-15:
Ok, thanks ;)

ralcaide, 2017-09-18:
Little bug, after restart, the RTC time is not correct 16-9-2013 05:04:28

fpgamcu, 2017-09-22:
TODO integrate NTP, since changing connection host and port is possible now https://github.com/Sinapse-Energia/AP-Sinapse/issues/47

ralcaide, 2017-10-11:
Time / date takes strange values like: 11-10-2017 24:18:48. The time should go from 00:00:00 to 23:59:59. 24:18:48 is not a valid time

fpgamcu, 2017-10-12:
The bug is related to UTC_OFFSET calculation. Also there is possible bug in AddSeconds() function. AddSeconds() not calculating month and years.

ralcaide, 2018-02-08:
Done and tested

closed 2018-02-08 by ralcaide