Sinapse-Energia/AP-Sinapse #43: [BUG - 713;] If there are a lot of devices in the NVM the 713; is cutted off

ralcaide, 2017-09-13:
Setting up 100 EPDs provokes that 313; returns a not well formed 713;. The last elements are not returned (buffer limit).

If the buffer limit is reached should the answer should be split, adding always the 713; at the beginning of the response.

ralcaide, 2017-09-13:
related with #15

fpgamcu, 2017-09-13:
This bug is related with `char	*CFrame::Build(char *dest){` context function.
There are 4 possible fix ways:
1) Increase `	static char	result[1024];` size from 1024 to for example 4096, i think it is not a best way
2) Split the response to several messages, will be not easy because need to redefine several things in engine to allowed split messages
3) Implement dynamic allocation, may be not so easy too because we can not easily know when to free the memory, maybe allocate when it needed and keep allocated for next messages, good because we can only add EPD and response to 313; can be only bigger with time, and I think several Kbyte of RAM will not be a lot (now about 1K for 100 EPDs).

Also if you need to test with more EPDs please check `MAX_TABLE_SIZE` define in context.h.

4) I can change definition of `static char	result` buffer size to be based on `MAX_TABLE_SIZE`, maybe it is good variant too

ralcaide, 2017-09-14:
Thanks for the analysis. 
The maximum number of EPDs to be managed will be 600. I think the best solutions are the 3 or the 4. The 3 will be maybe more efficient (memory) but I think the solution 4 will be more secure, isn't?. The solution 4 will occupy a maximum of 6-7 K, isn't?. 
What do you think?

fpgamcu, 2017-09-14:
Yes, solution 4 will use 7K max for 600 EPDs, I like solution 4 too because memory will me allocated statically, when in solution 3 allocation may fails when too many memory allocated in other routines.
Ok, I will implement solution 4.

ralcaide, 2017-09-14:
great, go for the solution 4.

ralcaide, 2017-09-14:
Ok, what is the limit?

fpgamcu, 2017-09-14:
I have tested with 200 400 and 600. 200 was ok

ralcaide, 2017-09-14:
Is the problem maybe related with the MQTT library? Maybe it is not supported very big messages. Have MQTT a limit for the size of the messages?. If it is like that will be necessary to modify the engine and this job will be done by other team, not by you. Please verify where is the problem and tell me

ralcaide, 2017-09-18:
To modify the implementation of the 313; command in order to return the EPDs by batches. Speak with API Engine Designer in order to find the best way.

Related issues : #15

ralcaide, 2018-05-07:
This issue is still not fixed and creating problems. Two fast improvements to the current behavior should be done:

1. Add the number of EPDs setup after the AP_ID
2. Cut off the response in order to not produce reconnections / HF in 100 EPDs

@fpgamcu, after these improvements, this issue could be set "To be tested". I think, the improvements are fast to develop. I set it to size = 2, please let me know if the estimation is not correct.

ralcaide, 2018-05-07:
To be developed in the current sprint

fpgamcu, 2018-05-07:
We need to modify context engine to have 1 implemented. Should I do this?
If implementing only 2 size of 2 should be enough

ralcaide, 2018-05-07:
Ok, I will ask the engine team. This task is on hold

ralcaide, 2018-05-08:
The engine have been modified in order to ease getting number of EPDs, and also enabling get and EPD by its position (Not by batches but only one). Attached patches files to be applied.

After this patches:

- Is it possible to add the number of EPDs to the 713; and limit the EPDs to be displayed to 100?
- If not, we will leave the 313; - 713; as it is and create a new command to just get the number of EPDs

[0002-JRB-114-Public-inline-Subspace-methods-ElementsCount.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1984349/0002-JRB-114-Public-inline-Subspace-methods-ElementsCount.patch.txt)
[0003-JRB-114-Correction-naming-to-previous-commit.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1984350/0003-JRB-114-Correction-naming-to-previous-commit.patch.txt)
[0004-jrb-114-Final-class-definition-3-public-methods-over.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1984351/0004-jrb-114-Final-class-definition-3-public-methods-over.patch.txt)
[0005-JRB-114-getElementbyIndex-implementation-plus-a-code.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1984352/0005-JRB-114-getElementbyIndex-implementation-plus-a-code.patch.txt)
[0006-JRB-114-CListSegment-Build-method-adjusted-to-use-ne.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1984353/0006-JRB-114-CListSegment-Build-method-adjusted-to-use-ne.patch.txt)

ralcaide, 2018-05-08:
@fpgamcu , please let me know if these changes ease the implementation of the required behavior

fpgamcu, 2018-05-10:
Yes, with this patches it is more easy to implement custom size 713; response.
One question to Engine team - is String variables in context somehow limited in size? if yes - what is the limit? I am asking because I will use context variable to store return value for 713;

ralcaide, 2018-05-10:
at first, there is no limitation. 
Question: why you want to save the string in a context variable? Could you not create the response without saving it in the context? The subparts are in fact in the context

fpgamcu, 2018-05-21:
>Question: why you want to save the string in a context variable? Could you not create the response without saving it in the context? The subparts are in fact in the context

Need to save answer as string in context because for now it is only possible way to pass something from engine as result of command.

ralcaide, 2018-05-22:
To be tested with C134

soporteHW, 2018-05-25:
Tested with T281: https://sinapseenergia.testrail.io/index.php?/tests/view/281&group_by=cases:section_id&group_order=asc&group_id=20

It doesn't published all EPDs that were setted up in the device, only from 100100 to 100191.
[image]

fpgamcu, 2018-05-25:
Can you please copy the response here as a text, not an image, to have characters count. I think response is limited by buffer size, and 100 is not enough limit. I will check.

fpgamcu, 2018-05-25:
For 100 EPDs if all EPD will have 3 digits for PNOM buffer should be something like (6+1+3+1)*100 = 11*100 = 1100, but we have only 1024 buffer for MQTT, I think it is possible to increase to 2048, I will increase and commit the code.

soporteHW, 2018-05-25:
713;E7178C;101;100100;100;100101;101;100102;102;100103;103;100104;104;100105;105;100106;106;100107;107;100108;108;100109;109;100110;110;100111;111;100112;112;100113;113;100114;114;100115;115;100116;116;100117;117;100118;118;100119;119;100120;120;100121;121;100122;122;100123;123;100124;124;100125;125;100126;126;100127;127;100128;128;100129;129;100130;130;100131;131;100132;132;100133;133;100134;134;100135;135;100136;136;100137;137;100138;138;100139;139;100140;140;100141;141;100142;142;100143;143;100144;144;100145;145;100146;146;100147;147;100148;148;100149;149;100150;150;100151;151;100152;152;100153;153;100154;154;100155;155;100156;156;100157;157;100158;158;100159;159;100160;160;100161;161;100162;162;100163;163;100164;164;100165;165;100166;166;100167;167;100168;168;100169;169;100170;170;100171;171;100172;172;100173;173;100174;174;100175;175;100176;176;100177;177;100178;178;100179;179;100180;180;100181;181;100182;182;100183;183;100184;184;100185;185;100186;186;100187;187;100188;188;100189;189;100190;190;100191;

fpgamcu, 2018-05-25:
Please, test with latest commit, once possible

closed 2019-03-28 by ralcaide