Sinapse-Energia/AP-Sinapse #44: [ML 5 - Create Frames] Develop a function that create a valid frame of each required command

ralcaide, 2017-09-13:
To develop the function (s) that generates any type of Sinapse RF Frame. This function is almost explained in [Create frames](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#2-create-frames)

**INPUTS**
- Command : PING, SET_STATUS, ASK_STATUS, SET_GROUP_ADDRESS and CONFIG_RADIO
- Destination address : 6 HEX characters
- SMART Bits : 2 Bits (Answer required & transit bit)
- Optional parameters : Some commands needs parameters, for example, SET_STATUS needs the dimming (from 0 to 100). The status (0/1) is directly from dimming. 

TTL always 06
Repeater (second smart bit) always 0
PA should be always incremented in +1, this should be done automatically inside of the create frame
Origin address is always 000001

**OUTPUT** 
- Frame : This frame will be used to create the burst of frames and save them in the queue in order to send them

**TEST**

In order to ease the testing, provide a method to call this function from MQTT and publish the result. For example:

Ask: 500;COMMAND;ADDRESS;SMARTBITS;PARAM1;PARAM2;PARAM4;PARAM4;
Response: 501;FRAME;

Example: Turn ON the EPD 200001

Ask: 500;SET_STATUS;200001;11;100;-1;-1;-1; (3 not needed parameters : -1)
Response: 501;0D 0C 00 00 01 20 00 01 FF 01 01 06 42;

fpgamcu, 2017-09-20:
Question about SET_STATUS command:
1. What format should receive `RF_FrameCreate` function for first parameter: voltage(0.0-10.0), percent(0-100), value (0-255) or hex value(00-FF)? (0-255 implemented in current commit)
2. Should `RF_FrameCreate` limit value of second parameter to 0-1? For example if I pass 2 as second argument 1 will be set to the frame. (don't limit in current commit)

ralcaide, 2017-09-20:
I try to respond your questions but please take into account the first point:

0. The RF_FrameCreate will be called only from the RF Part and the command 500; is only for testing purposes. So, the FW should know what kind of values should be passed.
1. The values of the parameters depend of the kind of command. See [commands & parameters](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#commands-for-ec). If the command is a "Set Group Address", the first parameter will represent the two last values (HEX) of the Group address and these value where received from the command 126;. If you receive a 126;FFFFAA;, the RF part should create a frame with the AA in the first parameter, if you convert the AA to a decimal value outside of the RF_FrameCreate you should convert again to Hex inside of the RF_FrameCreate. 
But take into account that a parameter is always mapped to a Byte. 
2. The second parameter will take the values 0 or 1 only of the command is "Set relays & dimming". In other commands to be developed in the future, the value of this parameter will have other values

Resume:

- A parameter represent always a byte, lets say that the parameters will be always a value between 0 and 255
- Inside of the RF_FrameCreate should be converted to a Hex value
- If any parameter have a not correct value for the particular command, then the frame will be not created. For example, the command "Set relays & dimming" can not receive some value > 1 as a second parameter. If receives something like that, then no frame will be generated and an error message should be published in the AP debug chanel

ralcaide, 2017-09-21:
TESTED OK - Creation of each kind of frame

ralcaide, 2017-09-21:
Description updated with: 
"RF: The generated frame should be also transmitted by RF. This will ease the testing"

@fpgamcu, please update the code in order to send the generated frame using TDMA. This is only for testing purposes

ralcaide, 2018-02-08:
Done and tested

closed 2018-02-08 by ralcaide