Sinapse-Energia/AP-Sinapse #47: [WIFI - HW - FW] To restart wifi module USR-232 after modify its configuration

ralcaide, 2017-09-18:
The WIFI module is working right but is not possible to restart it, so, after modify its configuration is not working as expected. To fix this problem is necessary to wait to the new HW revision.

Related issues that should be reviewed / retested after finish this issue:

#41 and #37 (Command 5;)

ralcaide, 2017-09-20:
Uploaded an updated doc of the USR-232 where is explained that is possible to restart the device using AT commands. See page 63. 

Command:
AT+Z

@fpgamcu, have you tested that?

ralcaide, 2017-09-24:
The last commit is provoking constant deafness in my device. After revert only the commit 3923fe6
in the last version of the code, everything works as expected. 
Should I set the USR in any special mode?. Should we set the USR through the web interface or just through the AT Commands?. Should I maybe modify the data set with the AT Commands?

fpgamcu, 2017-09-25:
I prepared a Demo video that shows my USR settings in web interface https://www.dropbox.com/s/dcwsj3w2a850cl5/Demo.wmv?dl=0. WiFi AP settings is set up using web interface. Server name is set up using AT commands. In video you can see that when I change server name with 5; command, it changes also in web interface automatically. Please also try my last commit, it may fix the problem (video is recorded before I made those last commit).

ralcaide, 2017-09-26:
I'm still having the same problem. I don't know why. Do you think is related with the web interface configuration?. I will try to get the current configuration

ralcaide, 2017-09-28:
It seems that now is working. After the commit https://github.com/Sinapse-Energia/AP-Sinapse/commit/12092fc6630eee4c22de66c8e3a03548877a5dc7

ralcaide, 2018-02-08:
Done and tested

closed 2018-02-08 by ralcaide