Sinapse-Energia/AP-Sinapse #49: [BUG - 1; and 3;] Acting over the AP with a non valid command for this device provokes strange behavior

ralcaide, 2017-09-18:
Acting over the AP instead of EPD with an EPD command provokes the action over the last EPD. This is not a correct behaviour, the command should be ignored.

> 2017-09-18 14:43:56 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:43:52> Incoming message #TESTING_AP/CMC/ACT/700099#:#1;#
2017-09-18 14:43:56 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 100;700099;200003;27;1;90;441.1;224.4;99.0;0.0;99;0.0;0;0.0;48;49362;
2017-09-18 14:44:20 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:12> Incoming message #TESTING_AP/CMC/ACT/700099#:#1;#
2017-09-18 14:44:20 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 100;700099;200003;27;1;90;441.1;224.4;99.0;0.0;99;0.0;0;0.0;48;49362;
2017-09-18 14:44:21 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:22> Start saving Device context to NVM
2017-09-18 14:44:22 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:22> End saving Device context to NVM
2017-09-18 14:44:29 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:22> Incoming message #TESTING_AP/CMC/ACT/700099#:#3;100;#
2017-09-18 14:44:40 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:32> Incoming message #TESTING_AP/CMC/ACT/700099#:#1;#
2017-09-18 14:44:40 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 100;700099;200003;28;1;100;491.3;223.8;110.0;0.0;110;0.0;0;0.0;48;53062;
2017-09-18 14:44:41 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:42> Starting Scheduler action
2017-09-18 14:44:41 INFO RX TOPIC:TESTING_AP/EPD/PERIODIC MSG: 100;700099;200001;23;1;50;159.8;218.9;35.0;0.0;35;0.0;0;0.0;48;51762;
2017-09-18 14:44:41 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <18-9-2017 14:44:42> Ending Scheduler action

ralcaide, 2018-02-08:
To be tested

ralcaide, 2018-04-25:
To be tested through C79

soporteHW, 2018-05-28:
Tested through C79 in 1.20 FW version. I sended the comand 1; and 3;X; a few times and always answer with 999;X;Wrong Topic; So it's OK.

closed 2018-10-29 by ralcaide