Sinapse-Energia/AP-Sinapse #51: [ML 6 - Reception of frames] To develop the reception of frames operations

ralcaide, 2017-09-19:
To develop the reception of the frames and its process as explained in the Wiki
[Reception of frames](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#reception-into-ap)
For this version / issue the AP should do basically the following:
- RX the frame
- Detect if it is a response and then save into the RX Queue
- Process the frame updating the needed context variables
- Update the RX and WR Queues

fpgamcu, 2017-09-25:
I have implemented RF receive code, but after send request I receive nothing. Please check my last commit RF_Tranceiver.c line 2520. Do I execute reception in the right way?

ralcaide, 2017-09-26:
After review the code we have the following remarks:

- gpio.c - line 282 - HAL_NVIC_SetPriority(EXTI0_IRQn, 5, 0); : Until now we have used always zero in the priority, (EXTI0_IRQn, 0, 0);. I don't know if this is creating any kind of problem
- main.c - line 337 - if (restartingTDA5340==1) : We think this line should be removed / commented

fpgamcu, 2017-09-26:
Thank you very much) Removing `if (restartingTDA5340==1)` helps.

ralcaide, 2017-09-26:
great!

ralcaide, 2018-02-08:
The reception should be improved but we already have other tasks to manage that. Clisng this one

closed 2018-02-08 by ralcaide