Sinapse-Energia/AP-Sinapse #53: [RF - Energy Consumed] To manage the energy consumed by an EPD

ralcaide, 2017-09-19:
TO Explain by RAE:

The EP and EQ should be calculated depending the status of the EPD and the Time. It will be updated periodically or after the reception of a RF frame related with the status

ralcaide, 2018-02-08:
To be designed

fpgamcu, 2018-02-23:
Hi, as I remember EP and EQ calculation was explained in other issue of AP-Sinapse. If it is possible, can you please point me to those issue number?

ralcaide, 2018-02-23:
We spoke about these parameters here: #34 and #35. It is also explained in the API but are not very well explained.

EP: Active Energy consumed by the EPD since set up in the context in Wh
EQ: Reactive Energy consumed by the EPD since set up in the context. With the current EPDs is not possible to know the EQ but we can assume EQ= 0,97 * EP. (To be improved in other issue)

EP: It is possible to calculate that using the P calculated using the I and V returned by the EPD. EP is it and addition of P per hour. If we know that the P of EPD was with 61W during 1 hour implies that EP = 61Wh during this hour.
Normally we have a P measure each several minutes so EP could be calculated:

EP(N) = P1 *(T1 - T0 / 60) + P2 (T2 - T1 / 60) + P3 * (T3 - T2 / 60)..... + PN * (TN - TN-1 / 60) = EP(N-1) + PN * (TN - TN-1 / 60)

Where T time in minutes since the device was set in the context.

To calculate that I think it is necessary to have the EP in the context and the time of the last measure.

ralcaide, 2018-03-30:
During a test related with the SD stability we found strange values on the energy:

`100;700004;520832;0;1;32;39.1;251.0;9.8;0.0;0;-845123.0;-819769;0.0;0;1353632982;`

Testing with version 5b5a7b3af5a9c4272975cf326050dfb48a7e26bf

Memory problem?

fpgamcu, 2018-03-30:
May be calculation problem. I will investigate more.

ralcaide, 2018-05-11:
@fpgamcu , any idea about negative numbers? Could you please take a look?

fpgamcu, 2018-05-15:
May be overflow, or wrong calculation. I will check.

ralcaide, 2018-08-21:
@fpgamcu , any feedback and estimation?

ralcaide, 2018-08-22:
This behaviour is creating serious problems. Any feedback? Do you know at least why is happening?

fpgamcu, 2018-08-22:
Unfortunately I can not reproduce this bug on my side. By the way, as I see you have not latest version of the binary, because decimal bug (-1.-05) and timestamp bug was fixed in latest commits.

ralcaide, 2018-08-22:
Right, is not the last version. We are still in testing phase.
Ok, We will finish testing phase and update the FW of these devices. Then we will see