Sinapse-Energia/AP-Sinapse #55: [ML 6 - Bootloader integration] To integrate the bootloader in AP working with ETH / WIFI

ralcaide, 2017-09-24:
See https://github.com/Sinapse-Energia/bootloader-stm32/issues/11

It is necessary to:
- Modify the client application (AP-Sinapse) to be saved in the correct position 
- Test that the behavior is correct
- Modify the wifi driver of the bootloader in order to be able to work using only AT commands (Same version as AP)

ralcaide, 2017-09-26:
This issue has high priority at this moment. We need that for tomorrow asap, around 12 will be great. Do you think is that possible?

fpgamcu, 2017-09-27:
I think it is possible and will do my best

fpgamcu, 2017-09-27:
Here is patch with Bootloader integration, sh
[0001-Change-start-address-and-linker-script-to-allow-debu.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1336959/0001-Change-start-address-and-linker-script-to-allow-debu.patch.txt)
ould I push it to git, or leave as patch?

francisjjp, 2017-09-27:
I am Francis from Sinapse Energia, Alvaro and Rafa colleague.

@fpgamcu, Have you modified repository for bootloader (bootloader-stm32)?. For downloading AP client program, it is needed to include one function at beginning of initialization  of bootloader program (maybe in Wifi232.c and Wifi232.h files into bootloader-stm32 repository) that allows to change the server for downloading.

The patch.txt is only for changing things inside AP client for allow the loading of own AP client program, but have you updated the (bootloader-stm32) repository. How have you tested this patch?

Thanks!

fpgamcu, 2017-09-27:
Updated patch 
[0001-Change-start-address-and-linker-script-to-allow-debu.patch.txt](https://github.com/Sinapse-Energia/AP-Sinapse/files/1337121/0001-Change-start-address-and-linker-script-to-allow-debu.patch.txt)
All those changes was pushed into repo

fpgamcu, 2017-09-27:
@francisjjp I am very sorry, saw your message just now. Yes, I have uploaded the code to bootloader repo. I have tested, seems AT code to change server to connect to is working ok on my board.

francisjjp, 2017-09-27:
Thanks a lot!.

francisjjp, 2017-09-27:
@fpgamcu . Two questions:

-  If you use one bootloader or you do not use one booloader. Will this program run?. I mean. I have seen that Flash origin of application program is now 0x8008000. Is this the entry program for both cases: with or without bootloader?. If so. Now, is it not needed to change anything for adapting application client?. is it?

- Another point. Where are the variables to use as the number of version for application program in order the bootloader can load one more recently application firmare?.

fpgamcu, 2017-09-27:
* No, theoretically if you have not bootloader this version of code shall not work
* Sorry, I don't know about that, shall I implement something more except of things in initial comment of these issue

francisjjp, 2017-09-27:
@fpgamcu, @ralcaide. 

We have tested bootloader with wifi, but it does not work. Regarding the AP client firmware. What is the revision that can be used without bootloader?.

francisjjp, 2017-09-27:
> Another point. Where are the variables to use as the number of version for application program in order the bootloader can load one more recently application firmare?.

Don't worry regarding this point.

ralcaide, 2017-09-28:
@fpgamcu , sorry for the misunderstandings. Please stay focus on the description of the issues. Could you please explain here the procedure that have you followed to test the bootloader?. Could you provide us with a client application with the AP_ID = 700091 and the TOPIC_ROOT = TESTING_AP in order to test in our side?

fpgamcu, 2017-09-28:
Yes, of course.
Testing steps: (since I was not sure how TrueStudio flashes the code, does it erases whole flash, I used STLinkUtility to flash the hex files)
1. Compile and flash bootloader into device
2. Compile and flash AP into device
3. Restart the device
Now bootloader is started, if it can not get firmware from server, it will start AP (after one or several minutes because of retries to connect)
Should I modify code to have those AP_ID and TOPIC_ROOT and push code to git?
Or I need to set up node with my device?
Should I change the AP_ID only in the AP code, or there is AP_ID somewhere in bootloader code too?

francisjjp, 2017-09-28:
Uploaded screen shots regarding Access Point USR-WIFI232-D2  working mode.

[AP Mode USR.pdf](https://github.com/Sinapse-Energia/AP-Sinapse/files/1340417/AP.Mode.USR.pdf)

francisjjp, 2017-10-23:
[6hours testing]. There are problems with bootloader ant last commit into AP (000456a commit).

1) AP client program without bootlader with FLASH_BASE = 0x8000000 and linker with FLASH = 0x800000 works good. It connects to MQTT broker.
2) AP client with bootloader with FLASH_BASE = 0x8008000 and linker with FLASH = 0x8008000, does not connect to MQTT broker. It seems AP client is loaded because USR-WIFI232-D2 changes network A to new NTP server included in this AP version.

Bootloader program has added the part of AT+TCPB=off, to avoid previous issues.

francisjjp, 2017-10-24:
@fpgamcu .

It is to need to test last version AP with bootloader commit (**b1353c284ffb471ba246a4430095b006fdd90d1a inside bootloader-stm32 project branch cortexM4**)

It seems AP client is loaded good (NTP server is loaded inte usr-wifi232-d2) but AP never connect to broker MQTT.

ralcaide, 2018-01-15:
TESTED OK

closed 2018-01-15 by ralcaide