Sinapse-Energia/AP-Sinapse #56: [ML6 - SEModule] To develop the extended commands to manage the SEModules

ralcaide, 2017-09-26:
The SEModules are other RF nodes that work a bit different in relation to the ECs. They share a big part of the commands but there are several differences. it is necessary to implement these differences as they are explained in the Wiki.

See [SEModules - RF](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#commands-for-semodule)

ralcaide, 2017-10-27:
These commands are not working as expected.
I'm performing a test with the device 100181 (SeModule) and the 1; provokes sending the command 08 (EasyControl) instead 02 + 11 (Ask Load in SeModule)

@fpgamcu , when are you sending the commands for the SeModules??

ralcaide, 2017-10-31:
After perform some testing it seems that all the commands for SE Modules are wrong:

**Status**
Sent: 0E A2 00 00 01 10 01 81 0C 00 00 B2 06 07
Expected: 0C A2 00 00 01 10 01 81 11 01 06 CS 

**ON**
Sent: 0C A2 00 00 01 10 01 81 08 C3 06 12
Expected: 0C A2 00 00 01 10 01 81 12 C3 06 CS

**OFF**
Sent:  0C A2 00 00 01 10 01 81 08 DA 06 29
Expected: 0C 02 00 00 01 10 01 81 13 DA 06 CS

**DIMMING**
The sent frame is also wrong.

See  https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#commands-for-semodule to understand how should work the SEModule commands.

@fpgamcu, could you have a look please?

ralcaide, 2017-10-31:
Testing:
- Now the Ask Load status is well sent and the response is received but NOT well managed. The Voltage and the current are not well extracted:
After ask about the status of a device we receives:
10 22 10 01 81 00 00 01 91 **01 E8 07 24** 09 06 79

01: Status ON -> OK
E8: Voltage after convert the E8 to Decimal -> 232V ( This value is not well extracted, the 100; contains 23.1 )
07: Current in A after convert it to Decimal and divide it between 60 ->  0,116 A -> 116 mA (This value is not well extracted, the 100; contains 2678.4)
24: Is the temperature after convert it to Decimal -> 24 (This value is not well extracted, the 100; contains a 3 as a temperature)
@fpgamcu , could you fix these little problems?

ralcaide, 2017-11-15:
With the last version, the RF commands are not good. When I'm asking to EC device like 523343, the AP is sending a SE module command (A2) and obviously the device is not responding?.

@fpgamcu, do know this problem?? Could you fix it, please?

I,m using the version 10887c7b89120288e41d9f83cdd812bfd88a336d compiled with USE_AP_CONTEXT

fpgamcu, 2017-11-15:
I added code that displays transmitted bytes.
Here is log that I have using version 10887c7 compiled with USE_AP_CONTEXT:
`TESTING/CMC/ACT/999988 313;`
`TESTING/CMC/DEBUG 713;`
`TESTING/CMC/ACT/999988 316;523343;10;`
`TESTING/CMC/ACT/999988 313;`
`TESTING/CMC/DEBUG 713;523343;10;`
`TESTING/999988/ACT/523343 1;`
`TESTING/CMC/DEBUG Node 999988 <15-11-2017 15:35:39> Will send 2 messages`
`TESTING/CMC/DEBUG Node 999988 <15-11-2017 15:35:39> RF TX 0B A8 00 00 01 52 33 43 00 06 82`
`TESTING/CMC/DEBUG Node 999988 <15-11-2017 15:35:40> RF TX 0B A8 00 00 01 52 33 43 01 06 83`
`TESTING/CMC/DEBUG Node 999988 <15-11-2017 15:35:41> Was sent 2 messages`
`TESTING/CMC/DEBUG Node 999988 <15-11-2017 15:35:41> Was recv 0 messages`
`TESTING/EPD/MEASUREMENTS 100;999988;523343;0;0;0;0.0;220.0;0.0;0.0;0;0.0;0;0.0;0;0;`
Seems all is ok.

ralcaide, 2017-11-15:
Strange because I see:
0C **A2** 00 00 01 52 33 43 **11** 16 06 A4 . 

This is the ask load command from SEModule, is this maybe related with a EEPROM problem? Now, my device is always sending SEModule commands instead of EC Commands

fpgamcu, 2017-11-16:
The problem is in definition of APIV in context. It is defined like hex 2 chars. So when in software we call SetVariable("EPD::APIV", "255") it can not set 255 (exactly set in one case and dont in other case randomly)
I will upload commit with fix APIV to integer type. also need to update context wiki.

ralcaide, 2018-02-08:
Done and tested.

closed 2018-02-08 by ralcaide