Sinapse-Energia/AP-Sinapse #58: [Bugs - Reception of the frames] Reception of the frames is not working well. Try to fix it

ralcaide, 2017-09-26:
Bugs found during the testing of #51 

Preliminary tests:

- The timestamp is not always correctly updated
- The values of the I and P are not correctly updated
- HF is produced sometimes

See log

> 017-09-26 23:57:49 INFO TX TOPIC:TESTING_AP/CMC/ACT/700099 MSG: 313;
2017-09-26 23:57:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> Incoming message #TESTING_AP/CMC/ACT/700099#:#313;#
2017-09-26 23:57:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: 713;523343;150;523343;150;523343;150;523343;150;523343;150;523343;150;523343;150;
2017-09-26 23:57:51 INFO TEST START: Basic actuation
2017-09-26 23:57:51 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:57:51 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:57:52 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;1;100;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31733;
2017-09-26 23:57:53 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> Start saving Device context to NVM
2017-09-26 23:57:53 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> End saving Device context to NVM
2017-09-26 23:57:53 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;50;
2017-09-26 23:57:53 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> Incoming message #TESTING_AP/700099/ACT/523343#:#3;50;#
2017-09-26 23:57:55 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:57:55 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:49:53> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:57:56 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 
100;700099;523343;0;1;50;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;

DSTAT OK - I KO - P KO - Timestamp OK

> 2017-09-26 23:57:57 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;80;
2017-09-26 23:57:57 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:03> Incoming message #TESTING_AP/700099/ACT/523343#:#3;80;#
2017-09-26 23:57:59 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:57:59 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:03> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:58:00 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;1;80;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;

DSTAT OK - I KO - P KO - Timestamp KO

> 2017-09-26 23:58:01 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;0;
2017-09-26 23:58:01 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:03> Incoming message #TESTING_AP/700099/ACT/523343#:#3;0;#
2017-09-26 23:58:03 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:58:03 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:03> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:58:04 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;0;0;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:05 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;150;
2017-09-26 23:58:05 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:03> Incoming message #TESTING_AP/700099/ACT/523343#:#3;150;#
2017-09-26 23:58:07 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:58:07 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Starting Scheduler action
2017-09-26 23:58:07 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Ending Scheduler action
2017-09-26 23:58:07 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:58:08 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;1;150;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:09 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;-10;
2017-09-26 23:58:09 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Incoming message #TESTING_AP/700099/ACT/523343#:#3;-10;#
2017-09-26 23:58:11 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:58:11 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:58:12 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;0;-10;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:13 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 3;100;
2017-09-26 23:58:13 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Incoming message #TESTING_AP/700099/ACT/523343#:#3;100;#
2017-09-26 23:58:15 INFO TX TOPIC:TESTING_AP/700099/ACT/523343 MSG: 1;
2017-09-26 23:58:15 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:13> Incoming message #TESTING_AP/700099/ACT/523343#:#1;#
2017-09-26 23:58:16 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;523343;0;1;100;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:17 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 1;
2017-09-26 23:58:17 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:23> Incoming message #TESTING_AP/700099/ACT/400030#:#1;#
2017-09-26 23:58:18 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;400030;0;1;100;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:19 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 3;50;
2017-09-26 23:58:19 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:23> Incoming message #TESTING_AP/700099/ACT/400030#:#3;50;#
2017-09-26 23:58:21 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 1;
2017-09-26 23:58:21 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:23> Incoming message #TESTING_AP/700099/ACT/400030#:#1;#
2017-09-26 23:58:22 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;400030;0;1;50;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:23 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 3;80;
2017-09-26 23:58:23 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:23> Incoming message #TESTING_AP/700099/ACT/400030#:#3;80;#
2017-09-26 23:58:25 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 1;
2017-09-26 23:58:25 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:23> Incoming message #TESTING_AP/700099/ACT/400030#:#1;#
[... 17 lines cut ...]
2017-09-26 23:58:39 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:43> Incoming message #TESTING_AP/700099/ACT/400030#:#3;100;#
2017-09-26 23:58:41 INFO TX TOPIC:TESTING_AP/700099/ACT/400030 MSG: 1;
2017-09-26 23:58:41 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:43> Incoming message #TESTING_AP/700099/ACT/400030#:#1;#
2017-09-26 23:58:42 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;400030;0;1;100;325.0;241.0;150.0;0.0;0;0.0;0;0.0;0;31793;
2017-09-26 23:58:43 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;
2017-09-26 23:58:43 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:43> Incoming message #TESTING_AP/700099/ACT/506070#:#1;#
2017-09-26 23:58:44 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;506070;0;1;100;1095.0;261.1;150.0;0.0;0;0.0;0;0.0;0;31843;
2017-09-26 23:58:45 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 3;50;
2017-09-26 23:58:45 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:43> Incoming message #TESTING_AP/700099/ACT/506070#:#3;50;#
2017-09-26 23:58:47 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;
2017-09-26 23:58:47 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:53> Incoming message #TESTING_AP/700099/ACT/506070#:#1;#
2017-09-26 23:58:48 INFO RX TOPIC:TESTING_AP/EPD/MEASUREMENTS MSG: 100;700099;506070;0;1;50;1095.0;261.1;150.0;0.0;0;0.0;0;0.0;0;31843;
2017-09-26 23:58:49 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 3;80;
2017-09-26 23:58:49 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:53> Incoming message #TESTING_AP/700099/ACT/506070#:#3;80;#
2017-09-26 23:58:51 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;
2017-09-26 23:58:51 INFO RX TOPIC:TESTING_AP/CMC/DEBUG MSG: Node 700099 <6-9-2013 08:50:53> Incoming message #TESTING_AP/700099/ACT/506070#:#1;#

****HARD FAULT****

2017-09-26 23:58:53 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 3;0;
2017-09-26 23:58:55 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;
2017-09-26 23:58:57 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 3;150;
2017-09-26 23:58:59 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;
2017-09-26 23:59:01 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 3;-10;
2017-09-26 23:59:03 INFO TX TOPIC:TESTING_AP/700099/ACT/506070 MSG: 1;

ralcaide, 2017-10-05:
Some comments / problems:

- When is updated the timestamp of an EPD?. It should be updated after each correct RF Response received. It is done like that?
- Now, it seems that the AP is working always with the same EPD Context but changes the EPD_ID. It is very strange. The behavior is a follows: Act over EPD_1, Ask about EPD_1 -> correct data of EPD_1. Then ask about EPD_2 -> Data of EPD_1 with with the EPD_2 ID. Have you saw this behaviour?

Please, test with the AP connected to my computer and the EPDs available in the NVM. Is this problem related with the problem of 313-713 that returns always the same EPD;PNOM?:

713;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;505047;30;

ralcaide, 2017-10-05:
It seems that the EEPROM of my device is not working well. Please ignore my comment by the moment.

ralcaide, 2017-11-22:
Question:

The timestamp of the responses is only updated when a RF RX related to the EPD arrives, right?. At least should be like that but I'm seeing an strange behaviour. The EPD is not sending any RF response because is disconnected but the messages that I'm receiving have the timestamp updated:

100;000007;100781;41;1;100;650.0;242.0;157.0;0.0;0;0.0;0;0.0;0;1342190900;
100;000007;100781;41;1;100;650.0;242.0;157.0;0.0;0;0.0;0;0.0;0;1342189616;
100;000007;100781;41;1;100;650.0;242.0;157.0;0.0;0;0.0;0;0.0;0;1342191153;
...
100;000007;100781;41;1;100;650.0;242.0;157.0;0.0;0;0.0;0;0.0;0;1342191926;

@fpgamcu , how is that possible? Could you confirm me that the timestamp of an EPD is only updated if a valid RF response is received?

ralcaide, 2017-11-28:
I tested it again and the timestamp is updated although the AP is not receiving any message related to the EPD.
@fpgamcu , could you take a look, please?

fpgamcu, 2017-11-30:
I have performed several tests:

For EPD that present timestamp is:
1343236664
1343236669
1343236675
(incrementing, ok)
1343236794
1343236815

For EPD that not present:
1342190903
1342190384
1342190648
1342191668
1342191670
1342189623
(abnormal)

And the same situation when i power off the present EPD.
When EPD not present RF recv code that change Timestamp at receive is not executed, So timestamp can not be changed, Seems like it is corrupted somewhere inside the engine.

ralcaide, 2017-11-30:
The timestamp is only updated in the RF recv??. If yes, this value should not be modified by the Engine in other point. 
Has the timestamp context var something special?

fpgamcu, 2017-11-30:
Updating only on RF packet received.
Should be regular variable, as any other.

ralcaide, 2017-11-30:
And the update is done after charge the new EPD, right?. So if it is like that I don't understand why is that happening

ralcaide, 2018-01-15:
TESTED OK

closed 2018-01-15 by ralcaide