Sinapse-Energia/AP-Sinapse #63: [ML 6 - I2C error] Failure with I2C EEPROM.

francisjjp, 2017-10-09:
There is a problem with EEPROM I2C. It seems problem is STM32F HAL libraries relative. It seems that this bug is registered under STMicroelecronics.  Some people tells it is a failure with HAL management.

General I2C STM32F issues.
1) https://community.st.com/thread/15632
2) https://community.st.com/thread/35884-cant-reset-i2c-in-stm32f407-to-release-i2c-lines
3) http://www.chibios.com/forum/viewtopic.php?t=560

Regarding shiffting I2C address. (IMPORTANT)

4) https://jure.tuta.si/?p=7
5) https://os.mbed.com/questions/55457/Is-there-are-bug-with-the-I2C-for-the-ST/

ralcaide, 2017-10-09:
@fpgamcu, what is the meaning of the last commit? This change could fix the EEPROM problem?

fpgamcu, 2017-10-09:
According to the https://community.st.com/thread/15632 it could. But I am not sure because can not reproduce problem on my device.

ralcaide, 2017-10-26:
The problem with the EEPROM still appears but is intermittent. We've added a question to the ST forum:

https://community.st.com/message/172945-problem-communication-stm32f405vg-and-i2c-eeprom-m24m02-drmn

@fpgamcu , is the problem well described? If you have more information, please add it here.

ralcaide, 2017-10-30:
Testing in other devices, the problem appears very often and does not disappears. We need urgent help / guidance on this issue

ralcaide, 2017-10-30:
@fpgamcu, Have you looked at this issue?: https://community.st.com/thread/35884-cant-reset-i2c-in-stm32f407-to-release-i2c-lines

What, do you think? Is it possible to try the solution proposed by the last comment?

ralcaide, 2017-10-30:
Also, we have found this post: https://electronics.stackexchange.com/questions/267972/i2c-busy-flag-strange-behaviour/281046#281046

What do you think? Is possible to try this solution also?

ralcaide, 2017-10-31:
One little thing:

We init the Pin WRITEPROTECTEEPROM_Pin but is not set to any value. @fpgamcu could you set it to 0, only for security.

ralcaide, 2017-11-13:
After check the HW, it seems that the problem is related with the FW. What is the status of this problem @fpgamcu

ralcaide, 2017-11-22:
Error catched in one device after 15 hours test. Attached snippet of the errors

[999994_I2CErrors.log](https://github.com/Sinapse-Energia/AP-Sinapse/files/1496349/999994_I2CErrors.log)

fpgamcu, 2017-11-22:
We received next debug data about i2c handler

hi2c1.Devaddress = 0 
hi2c1.ErrorCode = 32 // HAL_I2C_ERROR_TIMEOUT
hi2c1.EventCount = 0 
hi2c1.Init.AddressingMode = 4278190080 // corrupted
hi2c1.Init.ClockSpeed = 0 // corrupted
hi2c1.Init.DualAddressMode = 29685 // corrupted
hi2c1.Init.DutyCycle = 0 
hi2c1.Init.GeneralCallMode = 0 
hi2c1.Init.NoStretchMode = 0 
hi2c1.Init.OwnAddress1 = 0 
hi2c1.Init.OwnAddress2 = 0 
hi2c1.Instance->CCR = 0 // corrupted
hi2c1.Instance->CR1 = 537001984 
hi2c1.Instance->CR2 = 134290157 // corrupted
hi2c1.Instance->DR = 134275689 
hi2c1.Instance->OAR1 = 134275649 // corrupted
hi2c1.Instance->OAR2 = 134275653 // corrupted
hi2c1.Instance->SR1 = 134275691 
hi2c1.Instance->SR2 = 134275693 
hi2c1.Instance->TRISE = 0 // corrupted
hi2c1.Lock = 0 
hi2c1.MemaddSize = 0 
hi2c1.Memaddress = 0 
hi2c1.Mode = 64 
hi2c1.PreviousState = 0 
hi2c1.State = 32 
hi2c1.XferCount = 3 
hi2c1.XferOptions = 4294901760 
hi2c1.XferSize = 3 
hi2c1.pBuffPtr = 536929376 

Normally parameters should be

hi2c1.Devaddress = 0 
hi2c1.ErrorCode = 0
hi2c1.EventCount = 0 

hi2c1.Init according to line 62 from i2c.c
hi2c1.Init.AddressingMode = 16384
hi2c1.Init.ClockSpeed = 400000
hi2c1.Init.DualAddressMode = 0
hi2c1.Init.DutyCycle = 0 
hi2c1.Init.GeneralCallMode = 0 
hi2c1.Init.NoStretchMode = 0 
hi2c1.Init.OwnAddress1 = 0 
hi2c1.Init.OwnAddress2 = 0 

hi2c1.Instance->CCR = 32802
hi2c1.Instance->CR1 = can vary
hi2c1.Instance->CR2 = 41
hi2c1.Instance->DR = can vary
hi2c1.Instance->OAR1 = 16384
hi2c1.Instance->OAR2 = 0
hi2c1.Instance->SR1 = can vary
hi2c1.Instance->SR2 = can vary
hi2c1.Instance->TRISE = 13

hi2c1.Lock = can vary
hi2c1.MemaddSize = can vary
hi2c1.Memaddress = can vary
hi2c1.Mode = 64 // HAL_I2C_MODE_MEM, can vary
hi2c1.PreviousState = can vary
hi2c1.State = 32 // HAL_I2C_STATE_READY, can vary
hi2c1.XferCount = can vary
hi2c1.XferOptions = 4294901760 
hi2c1.XferSize = can vary
hi2c1.pBuffPtr = can vary

Seams some function access to memory where this handler stored and corrupts it. I'm not sure, but hardware reasons can also cause memory corruptions, but much rarely.
Also about power board. Today I have i2c errors without handler corruptions, right after board starts. This error stops when I disconnected power board, connected 5V, and next disconnected 5V and connected power board again.

But about current test log 99.5% that some function access memory by wrong pointer/out of borders array and corrupts i2c handler.

ralcaide, 2017-11-23:
Several questions related to the I2C handler:
- Is it possible to protect it somehow? Saving it in the MPU (if exists) or in any other place?
- The I2C handler is saved always in the same place? If yes, could we monitor to access to this part of the memory in order to detect what function is corrupting the handler?
- If the EEPROM problem is only related with the I2C corruption, it should be fixed after a restart, isn't? . I think this is not always happening . I will test it anyway

ralcaide, 2017-11-23:
Could we conclude that if the problem is fixed after restart is a FW problem and if not is a HW problem?

ralcaide, 2017-11-23:
Another thing: If the EEPROM problem appears the AP works very very slow and normally the command 314; is not executed. Is this fact indicative for something?

ralcaide, 2017-11-23:
After today meeting we agreed two approaches to do in parallel:
- Protect the pointer to I2C Handler. How? Creating a copy of the pointer in a secure place and test if the content is corrupted periodically. If yes, replace the current pointer for the copy.
- To research if it is possible to monitor a memory content in order to detect from where is the pointer modified

ralcaide, 2017-11-28:
It seems this bug fix solves the problem with the EEPROM.
Tested 3 devices during 17 Hours without problems. We will continue testing

ralcaide, 2018-01-15:
Main problem related with the FW fixed. Other problems found with HW. Closing this issue

closed 2018-01-15 by ralcaide