Sinapse-Energia/AP-Sinapse #64: [HW - EEPROM] Check schematics

ralcaide, 2017-11-02:
We are getting problems with the EEPROM, after a random time we get I2C busy. Please, check the schematics if everything looks fine
[EEPROM](https://github.com/Sinapse-Energia/AP-Sinapse/blob/master/Doc/commBoardV3p0_EEPROM_pageSchematic.pdf)

[MCU](https://github.com/Sinapse-Energia/AP-Sinapse/blob/master/Doc/commBoardV3p0_pageMicrocontrollerSchematic.pdf)

fpgamcu, 2017-11-02:
Strange thing. On schematic SCL and SDA resistors have 2.2k and on my board 4.7k

ralcaide, 2017-11-02:
Yes, true, this a mistake in the schematic. The resistors are 4.7k. Anyway, this difference should not affect.

ralcaide, 2017-11-02:
We've received feedback from ST engineer. He is asking about the write / read operations, are we waiting to end of the write operation to read and viceverse?. @fpgamcu, please, verify me this point.

Also, he recommend us to put the WP pin to GND in order to avoid problems. It is an Atmel recommendation.

ralcaide, 2018-02-08:
Closing the issue

redesign done and SD is now also available

closed 2018-02-08 by ralcaide