Sinapse-Energia/AP-Sinapse #65: [ML 8 - CMC - 1;] To develop the command 1; and its response 103;

ralcaide, 2017-11-03:
To develop the command 1; (On demmand) and 103;

[Ask about status on demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-about-status-on-demand)

ralcaide, 2018-01-15:
Assigned to Alex

ralcaide, 2018-01-17:
@fpgamcu , please provide feedback about the status of this issue. Could you provide an estimation about when we wiill have a first version? By the end of this week we should have something to test.

fpgamcu, 2018-01-21:
I have checked Alex code - there is bug with not working float formatting using sprintf. Alex will fix it at Monday.
Also there is a question. Where we should get Analyzed ID (MAC or IMEI) ?

ralcaide, 2018-01-22:
Thank you @fpgamcu,

Related to your question, it is the AP_ID as in any command. I've modified the wiki: [103;](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API) 

Could you provide also a little explanation about the code, please? We will use several analyzers, how could we add another analyzer? The interface for the FW should remain the same.

fpgamcu, 2018-01-23:
If you will use another analyzer you can modify either Analyzer.cpp getValue() function to read from different registers or create another analyzer class using Analyzer.cpp as a template and even use both analyzers simultaneously. Analyzer, MODBUS Protocol and RS485 objects are creating in the CreateContext() function.
But if you need just to switch to another analyzer, modifying Analyzer.cpp getValue() function will be enough.
If other analyzer will have different parameters to read, you should also modify command metadata in parametrization.cpp and Pre_REQAN() hook in wesouth.cpp.

ralcaide, 2018-01-30:
@fpgamcu , question:

Where is set the configuration of the analyzer? I see in the parametrization.cpp the baudrate configuration, is that all? In the analyzer we are setting other params like:

> Baudios: 19.200
Set CT2: 5
CT Rate: 0020
SyS: 3P4 (Tres Fases y Neutro)
Trafos: 100/5

ralcaide, 2018-01-30:
@fpgamcu , I saw the EASTRON analyzer has a maximum of 38400 Baudios. Why we have 115200 baudios by default in the code?

fpgamcu, 2018-01-30:
I checked the code. Seems Alex has not connected Context variable CMMODBBR with exact UART speed. Now Modbus UART speed is set up in usart.c line 260 (now using 9600). I will talk with Alex and he will link CMMODBBR with Modbus speed. The same thing is modbus address of Analyzed, maybe better will be to create another variable with this address (now using 1).

ralcaide, 2018-01-30:
Ok, any other thing related with the configuration of the analyzer that we should know?

fpgamcu, 2018-01-30:
For now there is no Analyzer configuration (Transformer setup, etc.) in current code. We tested this code with next Analyzer setup: Address 1, Baud rate 9600, Parity none, Stop bits 1

fpgamcu, 2018-01-31:
Context engine rounds decimal type values to one digit after point. So 0.98 is rounded to 1.0 and 0.93 to 0.9

ralcaide, 2018-01-31:
@fpgamcu , is necessary to modify the Engine, isn,t? If yes, tell me here I will plan the modification with the Engine team

fpgamcu, 2018-01-31:
Not necessary, we can change the variable type from decimal to string and save in any suitable for us format. Alex will do it today.

ralcaide, 2018-02-01:
ok, this issue is almost tested but we've found two little problems that should be fixed before close the issue:

- Use FDP as string in order to have at least two decimals precision
- Clean the code in order to have the modbus configuration parameters in the Context creation

To be done in the W6 Sprint if don't affect the development of the other planned issues. @fpgamcu , you decide that. If it is not possible will be planned for the W8 Sprint

ralcaide, 2018-04-25:
To be tested through C1

soporteHW, 2018-05-29:
Testing 1; command over a CMC, some errors/bugs was found:

1) USART2 baudrate must be defined as 9600 in MX_USART2_Init() function. Seems that line 612 in parametrizacion.cpp is not working fine.
2) Using a transparent sniffer, I see that when a 1; is set the following frame is sended:
  01 04 00 06 00 06 90 09  => It means: read with offset 00 06, the next 00 06 bytes (only read the Current parameters). Because of this, the Modbus analyzer response was: 01 04 0C 00 00 00 00 00 00 00 00 00 00 00 00 95 B7 and the MQTT response has all its params with value 0 (no lamps was connected, so it seems that the read values are OK). Why does this frame sended? the CMC should read all the analyzer registers, right?

Check this @ralcaide

ralcaide, 2018-06-07:
@fpgamcu , could you fix the above problems, please?

ralcaide, 2018-07-05:
Tested OK. T480

closed 2018-07-05 by ralcaide