Sinapse-Energia/AP-Sinapse #66: [ML 8 - CMC - Context] To update AP context in order to cover CMC requirements

ralcaide, 2017-11-03:
It is necessary to add several variables to the context in order to cover the CMC requirements. Please add it. The requirements are here:

[CMC requirements](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#inputs---subscriptions---management-of-the-cmc)

ralcaide, 2017-11-12:
Is this problem not related with some defined size in context.h?

fpgamcu, 2017-11-12:
I'm not sure, but seems like we just have not enough RAM to hold all context variables objects.
Is there any define that limit Device (non EPD) variables count?
And if it is, why this limitation causes HF and not just warning or compilation error?

ralcaide, 2017-11-13:
As it is explained in the WIKI, the AP is composed by an EasyNet and a CMC. It is possible to have the whole AP implementation but normally we use only a EasyNet or a CMC. Could we define what product we will use with a compilation directive or define?. If yes, this could be a work around to this problem by the moment, at least:

- AP : All the context vars 
- CMC : The EPD context vars are not necessary. These vars can be commented
- EasyNet : The CMC context vars are not necessary. These vars can be commented

fpgamcu, 2017-11-13:
Yes sure we can. I will implement this defines today

ralcaide, 2017-11-13:
Before implement, test if this workaround fix the problem, please

ralcaide, 2017-11-13:
After speak with the Engine team, they think that maybe is a heap / stack problem. It is possible that the heap memory is full. 
Do you have any idea about how much heap we have? Is it possible to increase it?

fpgamcu, 2017-11-13:
In linker file we define minimal size of heap. Maximum is limited by the size of free RAM. To increase size of heap we need to increase RAM size.

ralcaide, 2017-11-13:
The context vars are dynamically allocated, so they are in the heap, isn't? If the RAM is full and is necessary to create another context variable I understand that could appears a problem. But at the initialization, the RAM is free so I do not understand the problem. It should be enough RAM in order to create these new context vars, isn't?

Do you think the problem is related with the size of the RAM? Or is the problem maybe related with other fact?

fpgamcu, 2017-11-14:
> But at the initialization, the RAM is free

Yes, of course. If you execute code with this extra variables in context you will notice, that HF don't appear at start of program. Hard Fault exactly appears when context functions are executed. 
I debugged it some time and found that HF appears in line 150 of Frames.cpp.
In this line Frames for commands is dynamically allocated.
And it appears in both cases, when you add more variables, or when you add more commands. (When you add more variables - because of variables RAM is full and Frames can not be created.)

> It should be enough RAM in order to create these new context vars

If HF appear than there is not enough of RAM.

ralcaide, 2017-11-14:
I think our device has 192 MB of RAM, it should be enough. Anyway, have you implemented the workaround that I proposed you yesterday?

ralcaide, 2017-11-14:
The RAM memory is 196KBs (SRAM), are we able to know the size of the context in RAM?
In other hand, have you implemented the Workaround that I propose yesterday?

ralcaide, 2017-11-15:
CMC define tested : OK

ralcaide, 2018-01-25:
Done. By the moment the context are divided into CMC or EN.

ralcaide, 2018-04-25:
To be tested through C78

soporteHW, 2018-10-22:
Fixed. Related with  [#166](https://github.com/Sinapse-Energia/AP-Sinapse/issues/166)

closed 2018-10-22 by soporteHW