Sinapse-Energia/AP-Sinapse #67: [ML 8 - CMC - 4;] To develop the command 4; in order to public the status periodically

ralcaide, 2017-11-03:
See [Status periodically](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-about-status-periodically)

ralcaide, 2018-04-02:
@fpgamcu , question:

What happens if the EN receives: `4;30;22:00;06:00;-1`, will publish the status between 22:00 and 06:00 even if the hours are related to different days?

fpgamcu, 2018-04-02:
In current version it will never publish. Now condition of publish is next (nowAbsMinute >= startAbsMinute && nowAbsMinute <= endAbsMinute).
But it is possible to modify in order to publish between 22:00 and 06:00, if necessary.

ralcaide, 2018-04-03:
Thanks, an improvement is planned for next sprints. See #125

ralcaide, 2018-04-25:
Tested on T224

ralcaide, 2018-04-25:
To be tested through C9

ralcaide, 2018-05-14:
V1.18: Tested again with T224

closed 2019-03-28 by ralcaide