Sinapse-Energia/AP-Sinapse #68: [ML 8 - CMC - 11;] To develop the act over relays on demmand

ralcaide, 2017-11-03:
See [Act over relays on demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-about-status-on-demand)

ralcaide, 2017-11-12:
Ok, we will define the relays in Definitions.h. By the moment is enough with the first five relays

ralcaide, 2017-11-13:
@fpgamcu , are you able to define the relays 6 and 7, please? Do you need something from our side?

fpgamcu, 2017-11-13:
Yes, sure. I need MCU pin names where rellay outs is located

ralcaide, 2017-11-13:
@francisjjp , please provide the pin names of the relays 6 and 7

francisjjp, 2017-11-13:
Now we have following definitions.

```
#define CIRC_RELEGENERAL_Pin GPIO_PIN_6
#define CIRC_RELEGENERAL_GPIO_Port GPIOE
#define CIRC_RELE1_Pin GPIO_PIN_5
#define CIRC_RELE1_GPIO_Port GPIOE
#define CIRC_RELE2_Pin GPIO_PIN_4
#define CIRC_RELE2_GPIO_Port GPIOE

#define CIRC_RELE3_Pin GPIO_PIN_3
#define CIRC_RELE3_GPIO_Port GPIOE
#define CIRC_RELE4_Pin GPIO_PIN_2
#define CIRC_RELE4_GPIO_Port GPIOE
#define CIRC_RELE5_Pin GPIO_PIN_1
#define CIRC_RELE5_GPIO_Port GPIOE
```

The definition must be changed to this. (There are one swapping to "left" of label for relay numbers till 6th).

```
#define CIRC_RELE1 GPIO_PIN_6
#define CIRC_RELE1_GPIO_Port GPIOE
#define CIRC_RELE2_Pin GPIO_PIN_5
#define CIRC_RELE2_GPIO_Port GPIOE
#define CIRC_RELE3_Pin GPIO_PIN_4
#define CIRC_RELE3_GPIO_Port GPIOE

#define CIRC_RELE4_Pin GPIO_PIN_3
#define CIRC_RELE4_GPIO_Port GPIOE
#define CIRC_RELE5_Pin GPIO_PIN_2
#define CIRC_RELE5_GPIO_Port GPIOE
#define CIRC_RELE6_Pin GPIO_PIN_1
#define CIRC_RELE6_GPIO_Port GPIOE

#define CIRC_RELE7_Pin GPIO_PIN_15
#define CIRC_RELE7_GPIO_Port GPIOA
```
and inserting the extra code into MX_GPIO_Init in order to do work CLK for PA15, output mode,...

ralcaide, 2017-11-15:
Ok, @francisjjp, could you test it and give us the result, please

ralcaide, 2018-01-30:
Tested OK: https://sinapseenergia.testrail.io/index.php?/tests/view/7&group_by=cases:section_id&group_order=asc&group_id=1

soporteHW, 2018-05-17:
Issue reopened in order to be tested over new HW version.

ralcaide, 2018-05-17:
@soporteHW , please put here the ID of the test

soporteHW, 2018-05-30:
Tested in [TestRail - [T288]](https://sinapseenergia.testrail.io/index.php?/tests/view/288&group_by=cases:section_id&group_order=asc&group_id=29)

closed 2018-07-20 by ralcaide