Sinapse-Energia/AP-Sinapse #72: [ML8 - CMC - ASN - On Demmand] To develop command 305; and its response 705;

ralcaide, 2017-11-03:
See [ASN On demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-analog-input-value-on-demand)

ralcaide, 2017-11-13:
Question: What is the unit of the value of the response frame? for example, what is the meaning of the 1.210 in the following frame?

705;700099;2;1.210;-1;1341759549;

fpgamcu, 2017-11-13:
It's in Volt

ralcaide, 2017-11-13:
How is this feature developed in relation the steps and the values sent to back? What is the meaning of this 1.210?

fpgamcu, 2017-11-14:
It is voltage on your PA2 pin.
I changed ADC code now to use calibration register value. As i checked measured values is the same for both old and new ways, but new way is more correct ADC usage way.

> How is this feature developed in relation the steps and the values sent to back?
The steps next:
Receive 305;
Parse pin number, if it is not 2 - finish processing
Read ADC voltage on pin PA2
Prepare 705; answer
Put answer into MQTT out queue

fpgamcu, 2017-11-14:
For me message is 705;999988;2;3.096;-1;778054995;
I connected PA2 to RTC battery so 3.096 V is voltage on RTC battery.

ralcaide, 2017-11-15:
Asking several times with 305;2;, I'm receiving the following results:

705;700099;2;1.190;-1;1341947840;
705;700099;2;1.208;-1;1341947851;
705;700099;2;1.071;-1;1341947854;

The value oscillates between several values. 
@francisjjp, could you test with our sensors please and verify that everything is working right? If not, tell us the problems that have you found

ralcaide, 2018-04-25:
To be tested through C5