Sinapse-Energia/AP-Sinapse #74: [ML8 - CMC - ASN Threshold] To develop command 307; - ASN Threshold

ralcaide, 2017-11-03:
See [ASN Threshold](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-analog-input-value-threshold)

fpgamcu, 2018-01-26:
If I understand right 705; should be published NTIMES (-1 is forever) with granularity of 1 minute when ASN value is lower than LT or higher than HT. Should reaction time to LT and HT be equal to granularity? Or it should react faster?

ralcaide, 2018-01-26:
Ideally, should be faster if it is easy to implement. So, first time ASAP, then the other publish related with the same threshold should be each minute. 

It is that possible? If not, we leave the granularity to 1 minute alwys

fpgamcu, 2018-01-26:
I think there is some analog interrupt inside our MCU and we can react to LT and HT very fast using it. I will check.