Sinapse-Energia/AP-Sinapse #76: [ML 9] Migration of pinout

ralcaide, 2017-11-16:
Attached the differences of pinout between the old version of the HW and the new version. It is necessary to migrate the FW in a new branch

[Pinout_definition_v1p3.docx](https://github.com/Sinapse-Energia/AP-Sinapse/files/1478647/Pinout_definition_v1p3.docx)

ralcaide, 2017-11-27:
Thanks @fpgamcu,
is this migration finished? Could we already test with the new board using only EEPROM?

fpgamcu, 2017-11-27:
I think yes, but i am not sure that RF will work since i initialized it at start as hard SPI

ralcaide, 2017-11-27:
could you leave RF as a soft SPI in the meanwhile until the migration to hard SPI is done?

fpgamcu, 2017-11-27:
Sure, will change now

ralcaide, 2017-11-28:
We start with testing. Question:

How are you managing the reset pin of the gateway?. Now is wired and the gateway is restarted often

fpgamcu, 2017-11-28:
How this pin named in the table?

ralcaide, 2017-11-28:
I understand that is the PB0

@francisjjp , @soporteHW , could you verify, please?

ralcaide, 2017-11-28:
@fpgamcu, the pins:

USR_WIFI_RELOAD & USR_WIFI_RESET 

should be init to 1, if not, the USR Gateway is constantly restarting

ralcaide, 2017-12-06:
The migration is still not working, the device never connect to Internet because is blocked in the function SystemCoreClock_Config() function.

@fpgamcu , could you please work on this with high priority? We need to valid this new design ASAP.

Here you have a computer with a device connected and you can already debug:
Anydesk: alvaro-pc-125@ad // Sinapse2017

fpgamcu, 2017-12-07:
Can you share the schematic of new board?

ralcaide, 2017-12-07:
Sure:

[Eagle_EasynetCommboardV1p1.zip](https://github.com/Sinapse-Energia/AP-Sinapse/files/1538436/Eagle_EasynetCommboardV1p1.zip)

Please tell me if you need something more

fpgamcu, 2017-12-07:
According to this [table](http://www.txccrystal.com/crystal.html) load capacitance of our 26MHz crystal is 18pF.
Using formula from [here](https://community.nxp.com/thread/388856) we have next:
CX1 = CX2 = 2(18pF - 2pF) = 32pF or CX1 = CX2 = 2(18pF - 5pF) = 26pF
but on schematic we have 10pF, so crystal may just not start, but I am not sure.

Now I switched clock source to internal and it is not stuck on SystemCoreClock_Config()
But now can't connect to NTP or MQTT, seems like network config issue. I will proceed with this...

ralcaide, 2017-12-11:
Yes, you are right @fpgamcu. We have two crystal (internal and external) but only one RTC (internal). The RTC should use the external crystal in order to have more precision. By the moment, leave it as it is. I will create another issue with low priority in order to use the external crystal.

fpgamcu, 2017-12-14:
There is issue now on RF communication:
I am using our old RF code with soft SPI mode, hard SPI disabled and its pins configured as GPIO - exactly as in old code.
Starting of RF transceiver takes about 7-11 mins using function Transmit_RF_initialize() - holds for 7-11 mins on read_TEMP_INFINEON() function.

francisjjp, 2017-12-14:
@fpgamcu . Function read_TEMP_INFINEON() is not needed. Only for testing that TDA5340 goes right. Because the return value must be one value different to 0xFF. Maybe the problem is that initialitazing process for TDA5340  is very fast before to call read_TEMP_INFINEON()?

fpgamcu, 2017-12-14:
Maybe, strange that old board uses the same init code for RF.

ralcaide, 2017-12-18:
The EEPROM has not the necessary pull up resistors in place and this could be the reason of the bad behaviour.  
@fpgamcu , could you please set the internal MCU resistors to pull up and test if at least the EEPROM works?

ralcaide, 2017-12-18:
Commboard sent on Friday. Tracking UPS:

1Z0Y50F70498736126

ralcaide, 2017-12-19:
It is not possible to use internal resistors because the EEPROM Pins are set as a SPI instead of GPIO. We will change the commboard to include the resistors. By the moment, work only on the SD part

ralcaide, 2018-01-15:
@fpgamcu , could we close this issue? I understand that everything is migrated. One question:

- What about GPRS? Any migration performed? It should work?