Sinapse-Energia/AP-Sinapse #77: [ML 9] To use SD memory as a NVM

ralcaide, 2017-11-16:
EEPROM memory is provoking a lot of problems. The new version of the HW has an SD card. 
It is necessary to modify the FW in order to use the SD instead the EEPROM for the Context saving.

To test if the use of SD avoid problems

ralcaide, 2017-11-27:
I suppose this issue is not yet finished and we are still not able to test it, right?. Please update the status here: https://github.com/Sinapse-Energia/AP-Sinapse/projects/3
Put the issues into Done when we are able to test them.

Thanks

ralcaide, 2018-01-06:
Thanks, we will test this feature during the day. Are all the features developed or we wait more code?

fpgamcu, 2018-01-06:
Now EPD parameters are saving as text file and Device parameters are saving as binary. Will be good to save Device parameters as text also, but still it works with binary also.
Also I will test now with SDHC card, because I am not sure it will work like with SD.

fpgamcu, 2018-01-06:
I have tested with SDHC, works ok on my device.
One thing need to be done is 314; command adaptation to SD card.

ralcaide, 2018-01-06:
Great. The 314; should works transparently, yes

fpgamcu, 2018-01-06:
Unfortunately no, but I will modify it to call function like clearNVM(mode) and place it into NBInterface to separate implementation dependent code from other code.

ralcaide, 2018-01-08:
I'm not able to connect to Internet with this new version. Should I do something special?

fpgamcu, 2018-01-08:
I have tested on my board and all is working ok, including internet connection, tested with Ethernet and WiFi.
You can check several things:
1) If SD card is not inserted, or is not recognized - board will halt and blue LED will blink.
2) If blue LED is not blink - board should connect to the internet, please check your network configuration.

Is Blue LED blinking on your board after power on?
Also as I remembered on some of your board you have problem with starting RF for 15 minutes.
If you have remote access I can connect and debug.

ralcaide, 2018-01-08:
Thank you, now is working. It was a problem with the SD. It should be format in FAT32?

Please, push the code of the 314; when you have the new version

soporteHW, 2018-01-08:
Testing SD firmware version on a device with a 16 GB SD card, It works fine for 1 hour, After this time, the device doesn't publish mqtt messages anymore and NETWORK_LED blinks.

To fix it, SD card had been replaced for a new 8 GB SD card. Now, the device woks for 1 hour and 30 minutes. After this time, the device doesn't publish mqtt messages anymore and NETWORK_LED blinks.

I upload  the files loaded in last SD card: 
[SDfiles_EasyNetCommBoardV2_600095.zip](https://github.com/Sinapse-Energia/AP-Sinapse/files/1612428/SDfiles_EasyNetCommBoardV2_600095.zip)

fpgamcu, 2018-01-08:
I think it is related with a little amount of RAM that we have, but need to check. Files will not help a lot.
I will place my device in debug and put some breakpoints to catch this problem. But if I will not success in 2-3 hours can you provide the remote access to PC with your device so I will be able to place a breakpoint?

fpgamcu, 2018-01-08:
16 GB SDHC card, 3 hours, no problems yet. Seems problem is somewhere on your side. In any case I will leave active over night to check.

fpgamcu, 2018-01-08:
Problem appear also at my device, but after 5 hours of work. Seems like file system library can not write a file after some time. And the worst is that those one file that was corrupted stays on SD card with 0 size and blocks work on device in further. For now seems like a not enough RAM problems. I will debug more.

ralcaide, 2018-01-08:
Ok, yes, we have this file of 0 kbs also

ralcaide, 2018-01-10:
@fpgamcu , what is the status of this issue? Any advancement?

fpgamcu, 2018-01-10:
I have added additional debug output into the SD card operation and placed device to work several hours ago. I expect to have the debug output that will clarify the problem reason in 2 hours. Takes too many time to debug because issue appears not immediately but after some time of work.

fpgamcu, 2018-01-10:
After debug I get the next error on opening file for writing:
FR_DISK_ERR,			/* (1) A hard error occurred in the low level disk I/O layer */
Seems like hardware problem.

ralcaide, 2018-01-10:
Bad news, but this is a very vague error, isn't? Any idea about the possible reason?

ralcaide, 2018-01-10:
@fpgamcu , take a look please. Maybe is a FW problem.

http://community.openpicus.com/forum/software-ide-apps-and-libraries/6181/sd-card-error-1

ralcaide, 2018-01-10:
@fpgamcu , we will ask in the ST forum.
Could you provide us, please, the version of the FreeRTOS that we are using and the library that we use for the SD?

ralcaide, 2018-01-10:
After research a little bit in the ST Forum, we've found that sometimes there are problems with the RTOS and the SD writing / reading. In this thread there are some interesting ideas:

https://community.st.com/message/94511?commentID=94511#comment-94511

Also, it seems that the user https://community.st.com/people/Turvey.Clive.002 has a lot of knowledge related with SD cards. 

@fpgamcu , please:
- Check that we follow its recommendations (provided in the thread)
- Write a question explaining our specific problem given all the necessary details (RTOS, threads, writing flows, SD library, etc...)
-  Put here the link to the question. 

Thank you

soporteHW, 2018-01-10:
@fpgamcu If you dont't have a Log in register, you can use our registration:

user: soportehw@sinapseenergia.com
pass: napse1si

fpgamcu, 2018-01-11:
Hi, I have created a question on ST forum https://community.st.com/message/180823-fatfs-fopen-fails-with-frdiskerr-after-some-time
Also yesterday I was about to check if f_sync() helps, but MCU on new comboard get dead, so I will change it to other today and than continue.

ralcaide, 2018-01-11:
@fpgamcu , any idea about why the MCU get dead? This could be a big problem for us.

fpgamcu, 2018-01-11:
Seems like there was some solder residues on MCU pins, that shorted some pins and caused MCU damage.

ralcaide, 2018-01-11:
Strange because was working well during several days, isn't? Could you send us a photo?

ralcaide, 2018-01-11:
It seems that the code generated by Cube is creating some kind of problems:

https://community.st.com/docs/DOC-1751-wa-proposal-when-using-fatfs-r012c-with-stm32cubemx

The function of our code is not the proposed one. I will test it. 
What do you think, @fpgamcu ?

ralcaide, 2018-01-11:
The above workaround is using a function named SD_CheckStatus that is not available in our code. Are we using other version?

ralcaide, 2018-01-11:
Implemented the above workaround.
After some time a HF appears (maybe related with lack of RAM)
After restart the device the SD works but have lost several EPDs

ralcaide, 2018-01-15:
This issue is still in progress. The SD is not working after use it during several hours. It is necessary to work on that with high priority

ralcaide, 2018-01-17:
@fpgamcu , after finish with the bootloader, please continue with this issue.

ralcaide, 2018-02-05:
Testing:

01.02.18 - 16:00:00 : Test started
02.02.18 - 09:50:00 : OK - RF results very poor
05.02.18 - 11:00:00 : OK - RF results very poor. Waiting to new RF management queues

By the moment seems that the SD Driver is working good. Still testing the stability