Sinapse-Energia/AP-Sinapse #78: [ML 9] To use hard SPI instead soft SPI with RF transceiver

ralcaide, 2017-11-16:
In the new HW it is possible to use the hard SPI communication with the RF transceiver. Use it instead the soft SPI

ralcaide, 2018-01-26:
@fpgamcu , this issue will be finished in the current sprint? Today? If not we move it to next sprint

fpgamcu, 2018-01-26:
Will be finished today.

ralcaide, 2018-02-02:
@fpgamcu , what is the status of this task? Normally should be finished in the previous Sprint W4 but I think is not. I don't see any commit related with and also the results of the testing are very poor.

Could you provide me please feedback?

fpgamcu, 2018-02-05:
Sorry, There was some error in my local git repo. It displays me that all changes are pushed, but there is no changes on server. Now all pushed.

ralcaide, 2018-02-05:
Ok, thanks.
@fpgamcu , have you tested the RF with this new version? Should I see improvements or not yet?
The queue management algorithms have not been change, isn't? Was Alexander working on that several weeks ago, right?

fpgamcu, 2018-02-05:
We have found several issues in new queue implementation, that need to be fixed. In several hours Alex should finish testing of #69 and #92. After this he will be able to finish with queue management.

ralcaide, 2018-02-05:
OK, but I have some comments:

- Linked to what issue is done these queue improvement management?
- This bug fixing was not planned for the current Sprint but if Alex has time to do that after finish the planned work will be great. Please provide me the issue where is he working in order to insert it into the Sprint or to create a new one if it is needed.

fpgamcu, 2018-02-05:
I think we need new issue, something like "Improve RF communication to reduce amount of skipped incoming packets".

ralcaide, 2018-02-05:
Ok, here is the issue: #94 

Could I include it in the current Sprint?

fpgamcu, 2018-02-05:
Yes

ralcaide, 2018-04-25:
Tested OK

closed 2018-04-25 by ralcaide