Sinapse-Energia/AP-Sinapse #79: [Config] USR Wifi 232

sinapseenergia, 2017-11-17:
We are having troubles due to a bad configuration in this peripherical

It have normally activated DHCP Server mode, and this config obviously creates conflics into networks.

Config tags are not clear, it shows: "DHCP Server Enable" or "DHCP Disable"

What we need is DHCP Client Enable , DHCP Server Disable for Socket A

TODO:
Clarify DHCP correct configuration with vendor (USR)
Include automatic config from FW to avoid bad configurations

**Meanwhile, do not use DHCP, instead of this configuration , use an Static IP ALWAYS**

ralcaide, 2018-01-15:
To study status and design next steps.

ralcaide, 2018-02-08:
To clean the code and verify that all the options are working OK.
- Static IP
- DHCP