Sinapse-Energia/AP-Sinapse #81: [ML 8 - RF TX RX] To improve RX of the RF frames

ralcaide, 2017-11-28:
We are having very bad results related to the RF Reception and they could be related to the spectrum collapse because we are transmitting a lot of frames without waiting time. 
In order to do not collapse the spectrum and listen the most part of the RX Frames is necessary to slow down the TX in order to have less Responses and get the most of them.

See:[RF Transmission](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/RF-Sinapse-API#4-tx-of-the-frames)

ralcaide, 2017-12-14:
@fpgamcu , what is the meaning of this message:

100;000007;100780;0;0;0;0.0;255.0;0.0;0.0;0;0.0;0;0.0;0;1344405759;

RF answer received but I = 0 ?

fpgamcu, 2017-12-14:
if timestamp changed comparing previous than I think yes

ralcaide, 2018-04-25:
Tested OK

closed 2018-04-25 by ralcaide