Sinapse-Energia/AP-Sinapse #82: [ML 8 - RF] The command 126; is not sending the correct data

ralcaide, 2017-12-07:
The configure radio command is built with the following frame:

0C 16 00 00 01 00 00 02 RC 01 06 CS 

The value RC indicates if the EPD should be a master or not and the number of transmissions of a command is the device is a master. 
Now, is possible to set the EPD as a master but the number of transmissions is always zero. This is not correct. The number of transmissions should be set to 7.

So, now RC = 80 (1000 0000) but should be  87 (1000 0111)

fpgamcu, 2017-12-07:
Now I getting the number of transmissions from EPD::RFRT variable. Should I set it to 7 instead?

ralcaide, 2017-12-07:
How is set RFRT?

ralcaide, 2017-12-07:
This var should be at least different to zero, and by default should be 7

ralcaide, 2018-04-25:
To be tested through C77