Sinapse-Energia/AP-Sinapse #87: [EXTRA] Transparent mode for 3;X;

ralcaide, 2017-12-22:
Now, if an EPD is not in the context there is no possibility to act over it. This is OK if everything works right but we are having a lot of problems with the NVM and we should find a workaround in order to act over the devices when the NVM is not working.

To add a new mode with a define named TRANSPARENT_MODE. If this mode is ON, all the received commands 3;X; related with EPDs that are not in the NVM should send the associated RF burst.

With this feature, at least the down-link will work and we will able to act over the EPD but we will not have any data about the EPD status.

ralcaide, 2017-12-22:
@fpgamcu , what do you think about this issue? Is it simple to develop or you need a lot of work?. Please, provide me an estimation. This is an extra work but we need it till we have an stable NVM

fpgamcu, 2017-12-26:
Estimation is about 4h

ralcaide, 2017-12-26:
Ok, this is OK for us. When will you have time to do that? We need that before by 03.01.18. Is that possible?

ralcaide, 2018-01-15:
@fpgamcu , please integrate this development into Commboard V2.0 branch.

ralcaide, 2018-04-16:
Thanks for the implementations. After check the code I understand now is implemented only the command 3; as transparent, right?
What about to implement the 1; and 100; and maybe some other actuation over EPDs like Set Group address?. Do you think is worth and fast to implement? If not, maybe we can add a parallel command to 399;RF_FRAME_TX; in order to send using MQTT the received RF, something like 3999;RF_FRAME_RX;

fpgamcu, 2018-04-16:
I worked also at implementation of 1; command. At first I thinking of use our RF queues for receiving the answer. But after last commits it is not possible, because receive function now discarding messages with EPD IDs not in the context. So we should disable this feature for transparent mode (risking return of our bugs). Or there is another way, I thinking about to perform transparent receive outside queues (also need to freeze RF receive RTOS task because it can capture RF receive before transparent action and throw it away (because EPD not in context)). Estimated time to finish transparent mode is about 16h. Should also other commands be implemented in transparent mode, like 120; and 126; ?

ralcaide, 2018-04-16:
I understand the problems. By the moment we leave the transparent mode as it is developed right now. I will think / design the required features and will be developed in future sprints. 
At this moment the most important thing is the stability

ralcaide, 2018-04-25:
To be tested through C76

closed 2018-10-29 by ralcaide