Sinapse-Energia/AP-Sinapse #90: [EN V1.1 - LED] To improve the led behaviour in order to clarify where is the program

ralcaide, 2018-01-11:
The board has for LEDs but we are using only one with the next behaviour:

- If the board has power and the SD is working : Led ON
- If the board has power and the SD is not working: Led Blinking

This behaviour is not very useful. Develop the following:

- Led 1: OFF and the beginning, then blinking during bootloader working, ON when bootloader finishes and the go to Client App. Stays in ON during while the Client App is working.
- Led 2: OFF at the beginning and during the bootloader. Then blinking when the Client App starts and ON when is connected to the broker. Stays in ON when the app is connected to the broker. Blinking again if the connection is lost.
- Led 3: Displays the status of the EEPROM NVM. OFF at the beginning or if it is working with SD. ON if it is working right. Blinking if there is any problem.
- Led 4: Displays the status of the EEPROM SD. OFF at the beginning or if it is working with EEPROM. ON if it is working right. Blinking if there is any problem.

ralcaide, 2018-01-15:
The AP part is done. To implement the Bootloader part but in its project

sinapseenergia, 2018-01-25:
Looks like there is a BUG in LED expected behaviour.
Under NVM fault, device publish into DEBUG channel this problem, but in the tested case both LEDs remains ON
Please, check/clarify if in all cases which implies publish NVM fault in DEBUG channel, LED2 behaviour is changed to "Blinking"

ralcaide, 2018-01-26:
The previous behaviour was detected with EEPROM as a NVM

ralcaide, 2018-01-26:
Closing the issue because we will set only a RGB Led. It will be created a new issue to manage the behaviour of this Led.

closed 2018-01-26 by ralcaide