Sinapse-Energia/AP-Sinapse #91: [Compile for F427VG] Create needed files for compile the code for this MCU

ralcaide, 2018-01-24:
We are testing a F405VG compatible MCU with more RAM, named F427VG.
We understand that the code should be directly compatible and the only difference should be the files where the RAM is defined, .ld and stm32f427xx.h

Create / Modify the needed files in order to compile for this MCU

ralcaide, 2018-01-24:
@fpgamcu , could please give me an estimation of that? Is that easy?

fpgamcu, 2018-01-24:
An hour should be enough to perform migration.
I will also generate CubeMX project for it.

ralcaide, 2018-01-24:
Great. Should be managed through other branch? Could I include this issue in the sprint that is ending this week? It is only one hour so should not be a problem

fpgamcu, 2018-01-24:
Yes, sure. I can perform it today. Will you create new branch?

ralcaide, 2018-01-24:
Yes, branch:

https://github.com/Sinapse-Energia/AP-Sinapse/tree/Commboard-V2.0-F427VG

ralcaide, 2018-04-25:
Tested

closed 2018-04-25 by ralcaide