Sinapse-Energia/AP-Sinapse #92: [Set GPS Coordinates - 128;] To develop the command 128; for AP (EN + CMC)

ralcaide, 2018-01-26:
To develop the command 128; to set the GPS Coordinates of the device

[Set GPS Coordinates](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#set-gps-coordinates)

This command is valid for the whole AP. It is to say, it should be available in CMC and EN.

Related with #69 through sunset / sunrise

fpgamcu, 2018-02-05:
Should SR and SS be recalculated right after GPS coordinates changed by 121; command?

ralcaide, 2018-02-05:
Yes please, the SR & SS depends of the GPS Coordinates

ralcaide, 2018-04-25:
To be tested through T218

ralcaide, 2018-07-03:
After last modifications, the device is always restarted (Reboot!) when a 121; is received. This is not the required behaviour, maybe the TCP connection should be restarted but not the whole device.

@fpgamcu, what do you think? Can you please improve this behavior?

fpgamcu, 2018-07-30:
It is possible to implement only reconnect instead of reboot, but behavior will be the same. It will take the same time and will perform same operations. But reboot is more safer way of reconnection because it is also resetting state of communication driver. If it is possible, better will be to leave the current behavior, at least until #164 will be developed.

ralcaide, 2018-07-30:
Ok, thank you. We leave the current behaviour by the moment