Sinapse-Energia/AP-Sinapse #93: [Get Sunset / Sunrise, 204; - 604;] To develop the commands to retrieve the daily calculated sunset / sunrise time

ralcaide, 2018-01-26:
It is necessary to develop the actions and messages for calculating and retrieving the sunset / sunrise time based in the current coordinates of the device. 

Commands:

[Ask SS / SR](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#ask-sunrise--sunset-of-the-device)
[response SS / SR](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#sunset--sunrise-info)

The SS / SR should be automatically calculated once per day at 00:01

Task related with #69 and #92

fpgamcu, 2018-02-05:
If it is possible, it would be good to have next features added into the engine:
Time type, something that can hold hh:mm like values
Topics processing by the engine: analyzing and checking input topic and formatting output topic
Checking for commands format, for example - not react to 316; without epd id in it.

fpgamcu, 2018-02-05:
As alternative for creating new time type, is it possible to use ':' as variables separator, like ';', in commands metadata?

ralcaide, 2018-02-05:
Thank you for the feedback. There is already an Engine implementing the time type but is not integrated yet in the AP. I will plan this integration. Is totally necessary for the development of any issue?

fpgamcu, 2018-02-05:
Not necessary for now. 
If ':' as variables separator in metadata will work, than current issue is implemented completely, otherwise we can implement small workaround to make possible of HH:MM output.
By the way in current code we have similar workaround for parsing HH:MM as input variable.

ralcaide, 2018-02-05:
For this Sprint, please use the Workaround. Then, in next sprint will be planned the integration of new Engine version and the modification of the Workarounds, ok?

fpgamcu, 2018-02-05:
Sure

soporteHW, 2018-10-23:
Last test performed in FW V1.34

Test result: [121; & 204; Set GPS Coordinates & Ask about SR/SS](https://sinapseenergia.testrail.io/index.php?/tests/view/1012)

closed 2018-10-23 by soporteHW