Sinapse-Energia/AP-Sinapse #100: [Read MODBUS Generic - 311; - 711;] Read Modbus Generic on Demmand

ralcaide, 2018-02-09:
To implement Read Modbus Generic

[Read Modbus Generic] (https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-modbus---generic-on-demand)

ralcaide, 2018-04-25:
To be tested with C74

ralcaide, 2018-10-01:
Tested with bug (because bad design).
In the original description the modbus function is missed and then in the code is hardcoded to "04", that works good with a EASTRON analyzer but not with other modbus devices. The function modbus should be dynamic and provided in the MQTT command. 

@fpgamcu : See, please the new description of the command and adapt it.

ralcaide, 2018-10-02:
Improvement required:
INITIAL_POSITION now is coded in Decimal but we think is much better in Hex. 
The description of the command have been improved.

@fpgamcu , could you please take into account this new little requirement?

soporteHW, 2018-10-03:
The previous "311;" description was tested successfully.
Testt here:
[311 - Read MODBUS generic](https://sinapseenergia.testrail.io/index.php?/tests/view/1256)

soporteHW, 2018-10-16:
Tested with FW V1.40.1 => Tested OK
**NOTE** => in 711; response, the param NET_ANALYZER_CODE is always -1 value.

Test results here: [311; Read MODBUS - Generic (On demand)](https://sinapseenergia.testrail.io/index.php?/tests/view/1255)

However, with READ MODBUS GENERIC fixed, the command L3_READ_METERING_OD; fails. Maybe the bug is in interaction between L1-L3 functions. So, how L3 layer knows what type of net analyzer is connected and what function modbus is included?

fpgamcu, 2018-10-19:
As I remember we was implementing only one type of analyzer, and there is no logic now to select its type or use another analyzer.

fpgamcu, 2018-10-19:
That is why id was set to -1

ralcaide, 2018-10-25:
Command description updated removing network analyzer. Update the code please.

soporteHW, 2018-11-27:
Tested OK

Results here: [311; Read MODBUS - Generic (On demand)](https://sinapseenergia.testrail.io/index.php?/tests/view/1415)

closed 2018-11-27 by soporteHW

ralcaide, 2019-01-23:
Limitation: This command only works when the modbus devices have parity=none. 
TODO by RAE: Create a task to manage the modbus configuration (parity, bits, and speed)