Sinapse-Energia/AP-Sinapse #97: [Read DSN - 302; - 702;] Read DSN - On demmand

ralcaide, 2018-02-09:
To implement Read DSN on demmand

[Read DSN On Demmand](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-digital-input-status-on-demand)

fpgamcu, 2018-03-09:
Port number now can be for example B10 (PORTB pin 10 will be used) or just digits, 12 (PORTA pin 12 will be used). If it is incorrect behavior please tell me and I will fix it.

ralcaide, 2018-03-14:
Ok, I understand that A12 == 12, right? maybe in order to avoid misunderstandings we could use directly the number of the MCU PIN instead of the names. What do you think is a better option?

What PINs are now available as a Digital Input?

fpgamcu, 2018-03-14:
Using direct MCU PIN number is also possible, just need to define a pin table array with pin number - pin name assignment. it will take max 300 bytes of ram so should not be a problem.
Now with this function you can read any pin, even configured for output/af. Should I implement check code, to read only from specific pins?

ralcaide, 2018-03-14:
By the moment no, leave as it is. We will plan what to do through future issues. We the current version we can do what we need.

ralcaide, 2018-04-25:
Maybe this feature should be redesigned.

Waiting to decission