Sinapse-Energia/AP-Sinapse #99: [Read DSN - 304;] Read DSN on Change

ralcaide, 2018-02-09:
To develop Read DSN on Change

[Read DSN On Change](https://github.com/Sinapse-Energia/AP-Sinapse/wiki/MQTT-Sinapse-API#read-digital-input-status-on-change)

sinapseenergia, 2018-02-09:
In order to assure a better detection and avoid false changes due to a fast glitch, it´s neccesary to implement a little algorithm

Once a change (falling edge or rising edge) is detected on signal monitored by this way, it should provoques a interruption, but this condiction should not to be enought to trigguer the alert message

Once this happens, we should ask for DS value during X ms , a few times, and check if new value is stable. 

X should be customizable, in order to adjust to different kind of signals/enviroments

If this happens, then launch MQTT message

fpgamcu, 2018-03-12:
Should some specific pins be configurable for this command, or should any pin be configurable for this?

ralcaide, 2018-03-14:
Normally, it could be set as a digital input any GPIO not used for other purpose. What PINS are now configured as a digital input? These PINs are set in the CubeMX, isn't?

fpgamcu, 2018-03-14:
For other DSN related tasks there is no specific input pin configuration. Any input pin could be read using DSN on demand, even already used by other function ones.

fpgamcu, 2018-03-14:
For other DSN related tasks there is no specific input pin configuration. Any input pin could be read using DSN on demand, even those already used by other functions.