Sinapse-Energia/AP-Sinapse #103: [GPRS] GPRS communication is not working

ralcaide, 2018-02-15:
The GPRS integration has not been well done. Problems:

- CMC (Commboard 1.0): It connects but is reconnected each minute. The connection is totally unstable
- AP (Commboard 2.0): The Quectel M95 is not responding. The connection is never done

fpgamcu, 2018-02-15:
While performing debug found several important places in the code:
1) Cmdflow.cpp line 108
Delays specified here is sometimes critical. For example on my Commboard 1.0 first start is almost always ok, but after reconnection/reboot some commands response time is greater than 1 second. By the way, it will be better not to use this timing as delays, but use it as timeouts (if command will be received sooner we should not perform all the delay). But if connection timing is not critical - it is ok by now.
2) M95lite line 247
Here is specified timeout to read operation, it is now 10 sec and should be enough.
3) Deafness could be caused by the same reason that we had in USR driver - reading MQTT pub and ctrl received data into one queue. We was fixed this problem by next commit a02d5d63b700e323db51c5a0e25503aedc9d5d86

Deafness problem that we now have is not related with the integration, it is somewhere in new GPRS code. Most likely problem is somewhere in the DataBuffer management performing by the new code.

ralcaide, 2018-02-15:
Thanks.

1. Ok. By the moment increase please the delays in order to avoid problems
2. Ok. If it is necessary please increase it
3. We had the same deafness problem in the EC-M2M and we fixed it integrating the solution done by you with the DMA. Maybe after the two integrations there is a problem with the buffers. The DMA was never included in the GPRS part of the AP and maybe the DMA integration done in the EC-M2M is not compatible with the AP

fpgamcu, 2018-02-15:
On my Comboard 2.0 RES_MI_24_0402_1 was not soldered and SIM Card was not started. Please check that RES_MI_24_0402_1 soldered on your board.

ralcaide, 2018-02-15:
Thank you.

CMC : TESTED OK

Question: Do you connect using the command QLTS ?. This command always fail for me and normally it is not necessary in this product. The time can be get from NTP server. @fpgamcu , could you comment this AT command and get the time from NTP server as always?

fpgamcu, 2018-02-16:
For me QLTS is failed also.
> could you comment this AT command and get the time from NTP server as always?  

will do it

ralcaide, 2018-02-21:
@fpgamcu , question:
In Commboard 2.0, the pins powerkey and emerg_off should be set as OD (Open Drain). Is it like that?

francisjjp, 2018-02-21:
@fpgamcu 

Here are this specific schematic.

[image]

Yes, PWRKEY and EMERG_OFF must be set to OPEN_DRAIN and NOT PULL.

Then this part of code must be modified: M95_connect function:

	 	```
HAL_GPIO_WritePin(Emerg_GPIO_Port, Emerg_Pin, GPIO_PIN_RESET);  // This has changed because of OD
	 	HAL_Delay(400);
	 	HAL_GPIO_WritePin(Emerg_GPIO_Port, Emerg_Pin, GPIO_PIN_SET);  // This has changed because of OD
	 	HAL_GPIO_WritePin(Pwrkey_GPIO_Port, Pwrkey_Pin,GPIO_PIN_RESET); // This has changed because of OD
	 	countGPRSStatus = 0;

	 	do {
	 		HAL_Delay(2000);
	 		statusM95_statusPin = HAL_GPIO_ReadPin(M95Status_GPIO_Port, M95Status_Pin); //awaiting status pin goes to 1

	 		if (countGPRSStatus == nTimesMaximumFail_GPRS) { 
	 			HAL_GPIO_WritePin(Emerg_GPIO_Port, Emerg_Pin,GPIO_PIN_RESET);  // This has changed because of OD
	 			HAL_Delay(400);
	 			HAL_GPIO_WritePin(Emerg_GPIO_Port, Emerg_Pin, GPIO_PIN_SET);  // This has changed because of OD
	 			countGPRSStatus = 0;
	 		}
	 		countGPRSStatus++;
	 	} while (statusM95_statusPin == GPIO_PIN_RESET);

	 	HAL_GPIO_WritePin(Emerg_GPIO_Port, Emerg_Pin, GPIO_PIN_SET); // This has changed because of OD
	 	HAL_Delay(3000);
```

fpgamcu, 2018-02-22:
I just checked this. Yes changing GPRS init code to opposite (RESET on SET and SET on RESET) and swapping back PWRKEY and EMERG pins (PWRKEY to PD10 and EMERG to PD11) works as well as committed changes (PWRKEY to PD11 and EMERG to PD10). Sure that this assignment from last commit is wrong. I will commit proposed changes from last comment.

fpgamcu, 2018-02-22:
Also I checked OD and PP settings and seems it is not affect the situation, committed code will work as well if you will change settings of PWRKEY and EMERG from OD to PP.

ralcaide, 2018-02-26:
Last code is not working. The M95 is never initialized correctly. 
@fpgamcu , is working in your device? Could you please debug in our external 2?

ralcaide, 2018-04-25:
To be tested with C73