Sinapse-Energia/AP-Sinapse #105: [Time - RTC] Improve management of the current time of the device

ralcaide, 2018-02-23:
The time of the device is not very well managed. It is necessary to improve it. Now, that it is possible to set the GPS coordinates of the device it is possible to work with local time. Possibilities:

GPS coordinates are not set:
1. Get time from NTP in UTC, if NTP server is available
2. Time set by the user with the command 127;

GPS coordinates are set:
1. Get time from NTP in Local time, if NTP server is available
2. Time set by the user with the command 127; 

Considerations:

- The time should be recalculated each time that the GPS coordinates are set
- The time should be recalculated / corrected periodically if the RTC is not able to maintain the exact time

ralcaide, 2018-06-07:
The time management should be simplified and improved asap. The CMC is not usable with the current time behaviour.

@fpgamcu, please provide us feedback & estimation

fpgamcu, 2018-06-11:
Now the RTC is used in next way: after reconnection/reset time is got from NTP in UTC and set to the RTC, after used RTC to manage time.
I also noticed time drift and will check it reason.
Need to add, according to task description:
+ If the time is set by the user, then should be maintained after reconnections / restarts. So, should be managed by a flag in the NVM
+ Local offset: To be calculated and set when the GPS coordinates are set. If GPS coordinates are not set, then local offset is zero
+ If time is set by user (127;) the time is not synchronized.
+ If the time is coming from NTP, then the RTC should be refreshed automatically: Every hour and after reconnections / restarts
+ The offset should be recalculated each time that the GPS coordinates are set
+ The time should be recalculated / corrected periodically if the RTC is not able to maintain the exact time

Estimation is maximum 8 hours.

ralcaide, 2018-06-11:
This is a crucial task for the correct behavior of AP, CMC and EN. Please tell us if you find any difficulties or  doubts

ralcaide, 2018-06-17:
@fpgamcu , what is the status of this task? Are you facing any problems?

fpgamcu, 2018-06-18:
I have implemented it, in 2 hours I will push the code.

fpgamcu, 2018-06-18:
I noticed that we have not implemented 127; command. Should it be implemented in this task also.

ralcaide, 2018-06-18:
I think the 127; is implemented. At least it was tested with context AP. The command 127; should be available with any kind of context.

fpgamcu, 2018-06-18:
Yes, sorry, It was done outside the context.

fpgamcu, 2018-06-18:
Sorry for delay, there is several issues in current implementation:

Issues:
We have no offset management by now
Offset calculation by GPS coordinates is not done yet - databese of countries and timezones take several MB.
I will place databese on SD card and will try to use it, but not sure if it will work with such large files.

Limitations:
If each hour RTC correction by TIME server is enabled - device will reconnect each hour - it will take some time.
UTC offset now is integer in hours - that is why not possible to get UTC offset for timezone like +3:30

ralcaide, 2018-06-19:
So, if I understand, now we have, 
- RTC in UTC always
- No offset
- Synchronization, each hour. How much time normally need this reconnection?.

fpgamcu, 2018-06-19:
Depends of signal quality for GPRS (I think about a minute)
For ETH could be one or one and a half minute

ralcaide, 2018-06-19:
Thanks. Updated the description. The synchronization frequency should be configurable and we will set this in compilation time, each hour is not acceptable. We will use normally each 6 hours or so or we will design a MQTT to update the RTC

Related with the offset: To save the database in the SD is not a good idea. Maybe there is an online service to get the offset sending the GPS coordinates, isn't?. If not, we will make this service

fpgamcu, 2018-06-19:
Yes, there is several online services, but to work with them we should disconnect from MQTT before. If it is ok, I can implement it.

ralcaide, 2018-06-19:
Yes, the offset should be calculated only when the GPS coordinates are set. This fact normally will be done one once

fpgamcu, 2018-06-25:
Please check definitions.h file bottom for instructions on API key for UTC offset server.

ralcaide, 2018-06-26:
@fpgamcu, can I set the label "To be tested" in this task?

fpgamcu, 2018-06-26:
I think yes.
Offset calculation by GPS coordinates done using online service. By now offset still stored in the context as hours, but web service returns a seconds. So when we will implement offset management in #143 we could change offset in context to minutes to support time zones like India, etc.

ralcaide, 2018-06-26:
Ok thank you. I understand now offset is calculated but not used, right? everything related with time is in UTC.

fpgamcu, 2018-06-27:
yes

ralcaide, 2018-07-03:
Quesiton: If the time is set with 127;, how can be used again the NTP instead the user time? How is this flag reset?

ralcaide, 2018-07-03:
Problem:
after MQTT disconnection, the RTC have a big deviation.
[image]
@fpgamcu , please try to improve this behavior

ralcaide, 2018-07-03:
After three days of working, the RTC Time is not valid anymore.

[image]

What is the N by dafault?

ralcaide, 2018-07-05:
Another not valid behaviour:
- time set with 127; -> OK
- Disconnect the device at HH:MM
- After XX:YY
- Time expected: HH+XX:MM+YY
- Time of the Device: Very close to HH:MM

Is this problem maybe related with the RTC battery?

ralcaide, 2018-07-05:
@fpgamcu , all the problems related with RTC and time management have a very high priority. We will perform several modifications in the HW and this is the moment to detect any possible problem of possible improvement

fpgamcu, 2018-07-05:
May be battery issue. But may be also some issue in the code. I will debug it. 
Also I will add in 127; possibility to reset flag with -1 parameters.

By the way, about hardware modifications, I have several questions:
1) What crystals now connected to the STM32?
2) Is there some external RTC on board?
3) How battery is connected to the STM32?

ralcaide, 2018-07-05:
1. 
For MCU : 26 MHz
For MCU RTC: external crystal of 32768Hz

2. 
No

3. 
Connected to VBatt using Diode. We think the battery is well connected.

Questions:
- Are you using correctly the external crystal of 32Khz?
- Related with 127; after connection, are you getting the time from RTC with Get_RTC() or from Context?

fpgamcu, 2018-07-05:
Should be used correctly. Getting time from RTC. I will check also configuration of rtc, maybe there is an issue.

fpgamcu, 2018-07-11:
Have you measured battery current when testing battery circuit. When I measured I I got 79mA on 3.3V battery, it is huge. I am checking now SW side, could you share picture/pdf of battery circuit to check?

fpgamcu, 2018-07-11:
I have some schematic, but not sure it is latest. Here is the problem:
[image]
Leakage shown with the arrow

fpgamcu, 2018-07-11:
Also DIO_ME_3 not necessary here, and RES_MI_19_7 also, they can consume some current and reduce battery life, but better will be to speak with Eugene.

fpgamcu, 2018-07-11:
Also there was FW problem. Before last commit time was set on startup from backup registers, so there was effect of frozen time (time not tick in power off). Now I changed the code to use external crystal for RTC, not set time at startup - use time from RTC. Also on board I have removed RES_MI_19_2.

Checked with latest commit and removed RES_MI_19_2 - RTC is working when device is powered off.

ralcaide, 2018-07-11:
Thanks, with the RES_MI_19_2 was not working?

fpgamcu, 2018-07-11:
With RES_MI_19_2 battery current is 79mA so I think it was discharged by an hour or less.
Without this resistor current is about 1uA - 800nA so battery should live some time (need to calculate).

francisjjp, 2018-07-11:
@fpgamcu .

-  Yes you must remove RES_MI_19_2 in order not to discharge all battery on VDD. 
- RES_MI_19_7 is added , and it is 0 ohms, there is not consumption there. In next hardware review two diodes will be placed in this way: VDD + DIODE -> VBATT and BATTERY + DIODE -> VBAT.
-  Devices in sinapse has placed only DIO_ME_3 and RES_MI_19_7 right, RES_MI_19_2 is not placed (This bug in assembly was corrected round 1-2 months), but drift time continues.

soporteHW, 2018-08-02:
After several tests, it seems that RTC is working well.

With RTC configured through 127; API command, if device is power off for some time, after reconnection the device maintain a correct RTC value. So, the external battery is working.
See attached image:

[image]

soporteHW, 2018-08-02:
Although more stability tests will be performed, I included the label Status: tested passed.

ralcaide, 2018-10-29:
@soporteHW , could we close this task?

soporteHW, 2018-11-05:
Last test performed for EN device with FW V1.40.1

Test passed. Test result here: [RTC behaviour](https://sinapseenergia.testrail.io/index.php?/tests/view/1392)

closed 2018-11-05 by soporteHW